luxonis/depthai #305: segfault during calibration of OAK-D

albertoa, 2021-01-16:
OAK-D
Ubuntu 18.04
The installation of reqs and depthai are up to date as of today.
Connected via USB 2 port

Whether I use `./calibrate.py -s 2.4 ` or `./calibrate.py -s 2.4 -brd bw1098obc` the result is the same.

The calibration process starts and takes a picture of the chess grid. Eventually it segfaults. I'm concerned with all those write failures. Here is the output:

```
$ ./calibrate.py -s 2.4 
No calibration file. Using Calibration Defaults.
Depthai version required:   0.4.0.0
Compiled mode found: compiled for 7 shaves, 7 cmx_slices and 1 NN_engines 
Using Arguments= {'count': 1, 'square_size_cm': 2.4, 'image_op': 'modify', 'mode': ['capture', 'process'], 'config_overwrite': None, 'board': None, 'field_of_view': 71.86, 'baseline': 9.0, 'swap_lr': True, 'dev_debug': None, 'invert_v': False, 'invert_h': False}
Starting image capture. Press the [ESC] key to abort.
Will take 13 total images, 1 per each polygon.
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Usb speed : High/480Mbps
Mx serial id : 14442C10A193F4D000
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999733,   -0.022838,    0.003520,
    0.022827,    0.999735,    0.003021,
   -0.003588,   -0.002940,    0.999989,
  Rectification Rotation R2 (right):
    0.999603,   -0.028019,   -0.003088,
    0.028010,    0.999603,   -0.003024,
    0.003172,    0.002937,    0.999991,
  Calibration intrinsic matrix M1 (left):
  854.401672,    0.000000,  635.667603,
    0.000000,  854.906799,  407.493988,
    0.000000,    0.000000,    1.000000,
  Calibration intrinsic matrix M2 (right):
  856.665649,    0.000000,  646.718018,
    0.000000,  857.265198,  402.311462,
    0.000000,    0.000000,    1.000000,
  Calibration rotation matrix R:
    0.999964,    0.005164,    0.006775,
   -0.005204,    0.999969,    0.005858,
   -0.006744,   -0.005893,    0.999960,
  Calibration translation matrix T:
   -7.488523,
    0.209904,
    0.023137,
  Calibration Distortion Coeff d1 (Left):
   -5.105828,   16.132950,    0.000977,   -0.000418,  -15.642408,   -5.159005,   16.332483,
  -15.824655,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
  Calibration Distortion Coeff d2 (Right):
   -5.189439,   16.840321,    0.001261,   -0.000406,  -16.971752,   -5.239960,   17.031902,
  -17.152849,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
/home/alonsoa/projects/ml/luxonis/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json
depthai: Calibration file is not specified, will use default setting;
config_h2d json:
[... 369 lines cut ...]

E: [xLink] [    661819] [Scheduler00Thr] sendEvents:1027	Event sending failed
Closing stream right: DONE.
Thread for right finished.
Closing stream left: DONE.
Thread for left finished.
watchdog triggered 
=== New data in observer stream host_capture, size: 4
Writing 4 bytes to host_capture
E: [global] [    669415] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_WRITE_REQ

E: [xLink] [    669415] [Scheduler00Thr] sendEvents:1027	Event sending failed
!!! XLink write successful: host_capture (4)
Stopping threads: ...
Stopping threads: DONE 0.000s.
Closing all observer streams: ...
Closing stream host_capture: ...
Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    669415] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    669415] [Scheduler00Thr] sendEvents:1027	Event sending failed
Segmentation fault
```

alex-luxonis, 2021-01-17:
Hi @albertoa,

In rare cases with some hosts / USB hubs, we observed such instability over USB 2 connection, when running with the default USB3-enabled firmware. Switching to an USB2-only firmware resolved those issues.

I just pushed a branch [calibrate_add_usb2_opt](https://github.com/luxonis/depthai/compare/calibrate_add_usb2_opt) (commit https://github.com/luxonis/depthai/commit/388b022), on top of current `main`.
Could you try checking it out, and then run as: `./calibrate.py -s 2.4 -brd bw1098obc -fusb2`

albertoa, 2021-01-17:
Hi Alex, thank you for the really quick response.

I was wondering about it since I saw an indication of port issues in the troubleshooting page, but my problem didn't quite match what was described there.

The new branch with the -fusb2 flag works great. No more crashes during calibration.

Thanks,

Alberto

closed 2021-01-17 by albertoa