luxonis/depthai #306: Running setup example fails when using a path that has a space in it

charlesfracchia, 2021-01-17:
On Ubuntu 20.04, following the setup/getting started tutorial failed for me when attempting to run the demo script `python3 depthai_demo.py`. After taking a quick look at the error that was thrown (thrown by [this line](https://github.com/luxonis/depthai/blob/3b277fafe4ac09768d017ed61e0624d876c99d73/model_compiler/model_compiler.py#L30)), I quickly realized that my working path had a space in it, and that it was breaking along the way.

For expediency, I changed the path to remove the space, but this should probably be fixed for newcomers :)

closed 2021-01-17 by szabi-luxonis

Luxonis-Brandon, 2021-01-19:
Adding 

RuntimeError: Model downloader failed! 

for google-ability.

samgutentag, 2021-01-27:
ran into the same problem with a '-' hyphen in my virutalenv name, example the virtualenv "oak-d" did not threw the same "Model downloader failed!" error

switching to a virutalenv called "oakd" and now it works! not sure if that was the problem, but it was a fix for me

running on Raspberry Pi 4b

Luxonis-Brandon, 2021-01-27:
Oh that's super good to know.  We had run into weird virtualenv things before and hadn't resolved them.  I bet it was the `-` hyphen!  Thanks for the information here!

PR #307 (2021-01-17): Fix error when there is space in path
Fixes #306 
Contains fixes from : #272