luxonis/depthai #310: Is is possible to provide API and lib for Java ?

wureka, 2021-01-19:
I am a Java developer.
Is is possible to provide API and lib for Java ?
Many thanks.

Luxonis-Brandon, 2021-01-19:
So we do have a community member who is working on this.  I'm asking him if I can put you two in touch.  The C++ bindings should be fairly straightforward to use in Java.  https://www.codeproject.com/Articles/993067/Calling-Java-from-Cplusplus-with-JNI#:~:text=The%20Java%20Native%20Interface%20(JNI,Java%20components%20from%20C%2B%2B.

I think this member already got this working though, so it would reduce work duplication.  So I just pinged him.

Luxonis-Brandon, 2021-01-19:
I also just made a channel in the Discord for this:
https://discord.com/channels/790680891252932659/801236927080038430/801237086069850132

Invite link:
https://discord.gg/EPsZHkg9Nx

Grumpy141, 2021-01-20:
Hi, yes I have started, but it is not working yet.   I decided to create the JNI interface using Javacpp.   It parses all the C++ header files and creates the JNI and jar files.  I got all the parsing to work, but it creates a 7000 line c++ file that must compile for the JNI.   I am erroring out at line 5000, so it is close.   I have no c++ experience, so it is slow going.  If there is a Java developer who has some c++ experience, could probably get this working in a day.

Luxonis-Brandon, 2021-01-20:
So pinging @jonngai on this actually.

@jonngai - would you be able to help.  It just struck me that with your Android experience you may be able to swoop in and help out.  

Thoughts?

jonngai, 2021-01-20:
Hi @Grumpy141 , yeah, I might be able to help out a bit as time permits. Is there any way you can share what you've got so far and where you're getting stuck?

Grumpy141, 2021-01-20:
Hi,   Please send me your e-mail address and I will share with you what I have so far.   My email is grumpy141@yahoo.com.Thanks

wureka, 2021-01-20:
Hi  @Grumpy141,
Thank you for your sharing. My email is wureka@gmail.com
My main developing language is Java.
However OpenCV AI Kit currently only provides python API.
So, I am thinking of accessing depthai-core libraries with Java JNA.
But unfortunately, I am not familiar with C++, nor JNA. So, it's a big
challenge for me.
Anyway, let's see what I can contribute.

Thank you again.
Regards

Grumpy141, 2021-01-21:
Hi @wureka,   I am working on the JNI at the moment.  Once we have that, you could help with the Java classes for the Java API.  
  To create the JNI, I am using javacpp.    bytedeco/javacpp

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
bytedeco/javacpp

The missing bridge between Java and native C++. Contribute to bytedeco/javacpp development by creating an accoun...
 |

 |

 |

I am creating a new javacpp-preset called depthai.  You can find the javacpp-preset here:
bytedeco/javacpp-presets

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
bytedeco/javacpp-presets

The missing Java distribution of native C++ libraries - bytedeco/javacpp-presets
 |

 |

 |

You won't need any of these once the Java interface is created.   javacpp parses all the C++ header files and automatically creates the JNI interface.   If you want to help with the JNI interface, then download the javacpp-presets and get one of the individual projects to compile on your machine.  There is a wiki for both javacpp-presets and javacpp.   Or you can wait until the JNI interface is created and then start chipping in.Thanks

themarpe, 2021-09-08:
With JavaCPP preset for `depthai-core` released over at https://github.com/bytedeco/javacpp-presets/tree/master/depthai will be closing this issue. Thanks everyone for their input:)

closed 2021-09-08 by themarpe