luxonis/depthai #311: Triage: Jetson install-depencies warning.

eric-schleicher, 2021-01-20:
Opening issue to triage warning that occur on Jetson (nano) Jetpack 4.4 since chasing another issue, starting at the beginning, will close if validate complete install process.

Full output [here](https://gist.github.com/eric-schleicher/6729d2c51535e1c770b7005eb6bf1d37)  but the part that looked a bit concerning was: 

```
...
Processing triggers for initramfs-tools (0.130ubuntu3.9) ...
update-initramfs: Generating /boot/initrd.img-4.9.140-tegra
Warning: couldn't identify filesystem type for fsck hook, ignoring.
I: The initramfs will attempt to resume from /dev/zram3
I: (UUID=11ba3b3c-cb8f-46ec-8b69-f3b1d04a1421)
I: Set the RESUME variable to override this.
/sbin/ldconfig.real: Warning: ignoring configuration file that cannot be opened: /etc/ld.so.conf.d/aarch64-linux-gnu_EGL.conf: No such file or directory
/sbin/ldconfig.real: Warning: ignoring configuration file that cannot be opened: /etc/ld.so.conf.d/aarch64-linux-gnu_GL.conf: No such file or directory
...  *(completed without error)*
```

this appear to point at different file name or location than expected because of tegra/jetpack FS.

Luxonis-Brandon, 2021-01-20:
So Jetson series have their own OpenCV build.  So we need to modify that script (and any other scripts that are run on Jetson that install dependencies) to not try at all to install OpenCV, as it is already pre-installed on JEtson and trying to install another version of OpenCV will likely break a lot of things.  

So for installing on Jetson, the best approach is likely to skip all the dependencies, and literally just install depthai itself.

At least until we can make scripts that are targeted for the Jetson.

Thoughts?

eric-schleicher, 2021-01-20:
It's curious that they have their own build...  because...  for some totally **iNSANE** reason the openCV build on the default jetpacks aren't even CUDA enabled
[image]

While it's not anyone other than Nvidias job to address that, making jetson setup deliberate and high probability of success benefits everyone.

AFAIK There are two known good in jetsonland solutions for this.

https://github.com/JetsonHacksNano/buildOpenCV
(popular/canonical)

and 
https://github.com/mdegans/nano_build_opencv
(flexible)

If this were the case, would the *right* files be in an expected (or even just known) location?

eric-schleicher, 2021-02-19:
looks like this is resolved.

closed 2021-02-19 by eric-schleicher