luxonis/depthai #313: [BUG] Online blob converter generates generates a .bin file instead of a .blob file

geaxgx, 2021-01-20:
Hi,
I have used the Online blob converter to do the 2 steps conversion from a Tensorflow .pb file to a MyriadX blob file.
The generated file by the tool was suffixed '.bin' instead of '.blob'.
After the download, I changed the suffix to .blob, and it worked, so it means the generated file was really a .blob, but just mis-suffixed.

VanDavv, 2021-01-20:
Thanks for the report @geaxgx! I also reproduced it and will fix shortly

geaxgx, 2021-01-24:
@VanDavv  I have discover another small issue with the online converter. With Openvino 2021.1 and if using the default MyriadX Compile params, I get the following error: ERROR: unknown command line flag 'VPU_MYRIAD_PLATFORM'

VanDavv, 2021-01-25:
@geaxgx Fixed both of the issues, they should be already deployed in blob converter - http://69.164.214.171:8083/

closed 2021-01-25 by VanDavv