luxonis/depthai #319: [BUG]

nlapenn, 2021-01-24:
**Check if issue already exists**
Note:  Troubleshooting has something that may be related (Pi's can lock up due to power consumption).  This may be the same issue but it happens with the Oak-D running on its own power so I would hope this would alleviate power draw issues.

**Describe the bug**
Raspberry Pi 3 b+ hangs while running 04_rgb_full_resolution example from depthai gen 2 with Oak-D and Oak-1

**To Reproduce**
Steps to reproduce the behavior:
1. Run script:  examples/14_color_camera_control.py
2. Use the demo for 5-30 seconds
3. System hangs.  No error message.  Have to power cycle to reset the system.

**Expected behavior**
Example works and doesn't hang.  Alternatively, an error message and elegant failure.

**Attach system log**
[log_system_information.zip](https://github.com/luxonis/depthai/files/5863132/log_system_information.zip)

**Additional context**
I also have an adafruit MotorHat which is also running on its own power supply - removing it and all other peripherals didn't change the behavior.

I've also reproduced the issue with 04_rgb_full_resolution_saver.py

Other notes:  the system log came from the gen 1 pipeline as I couldn't find a log_system_information in gen 2.

Using gen2 pipeline:
Repo:  http://github.com/luxonis/depthai-python.git
SHA: 2eb6639f178a928f6b716e7b6ff18df9b2ee66b8 gen2_develop

nlapenn, 2021-01-24:
It looks like on 04_rgb_full_resolution_saver.py,  adding usb2Mode = True when initializing the device prevents the hang.  Unfortunately it didn't help with 14_color_camera_control.py

szabi-luxonis, 2021-01-24:
> It looks like on 04_rgb_full_resolution_saver.py, adding usb2Mode = True when initializing the device prevents the hang. Unfortunately it didn't help with 14_color_camera_control.py

Are you using the usb cable that came with the device?

nlapenn, 2021-01-24:
> 
> 
> > It looks like on 04_rgb_full_resolution_saver.py, adding usb2Mode = True when initializing the device prevents the hang. Unfortunately it didn't help with 14_color_camera_control.py
> 
> Are you using the usb cable that came with the device?

Yes.

Luxonis-Brian, 2021-01-25:
Adding to discussion in case it's relevant: the RPI 3B+ only has USB 2 ports. Shouldn't need to set usb2Mode = True. Perhaps an issue with the negotiations? Or maybe can be reproduced on a different USB 2 port?

Luxonis-Brandon, 2021-01-26:
@nlapenn - so the full resolution saver will use a lot of resources on the Pi.  I suspect the Pi 3B+ is actually hitting an out of memory (OOM) condition.

Can you run `top` while trying this on the Pi 3B+?  (Or otherwise monitor memory?). I bet the Pi is running out of memory or perhaps other resources (writing to fast to disk, choking) as a result.

We should clarify that some of these tutorials are high-data-rate and should be throttled if run on a Pi 3B+ or similar host.

Thoughts?

Thanks,
Brandon

VanDavv, 2021-01-26:
I've reproduced this error on this [video](https://youtu.be/IuHjy2ZsEyc) (with `htop` enabled), but it seems we have the same issue also when building depthai-python library on RPI 3B+, same crash happens - I've recorded this situation on this [video](https://youtu.be/EBBzhyknaIA) (CC @themarpe @szabi-luxonis )

themarpe, 2021-01-26:
@VanDavv this is as @Luxonis-Brandon mentioned, out of memory issue.
For building, non parallel build (or max 2 thread build) should be used with this amounts of RAM.
As for the example, queue sizes should be checked - the host side queues fill up with all the data.
It seems that currently we queue up to 30 NV12 frames which at 4K resolution equal to ~356MiB. 
@VanDavv can you recheck on some other host, if memory consumption keeps on growing?

nlapenn, 2021-01-26:
> @nlapenn - so the full resolution saver will use a lot of resources on the Pi. I suspect the Pi 3B+ is actually hitting an out of memory (OOM) condition.
>
> Can you run `top` while trying this on the [...]

I've attached a screenshot with the output of top when the system hung

[image]

Also, I apologize - I swapped back and forth in the original report between 04_rgb_full_resolution and 14_color_camera_control.  Just to clarify - they both hang the system.

Luxonis-Brandon, 2021-01-26:
Thanks for the update.  We're investigating internally.  Trying to figure out what exactly causes the Pi to hang here.