luxonis/depthai #321: 3D Scan

Zumbalamambo, 2021-01-24:
I have purchased OAK-D. I would like to know if it's possible to create a 3D scan of the environment and retrieve the depth images, RGB images and their corresponding camera poses?

Luxonis-Brandon, 2021-01-26:
Great question @Zumbalamambo and sorry about the delay.  So actually many users are now working on integrating OAK/depthai with SLAM solutions.  

Below is a summary of some of the current in-flight efforts:

- https://github.com/ColbyToland/oak_slam
- https://github.com/duncanrhamill/oakd_orbslam3
- https://www.youtube.com/watch?v=3b_P6yItYxM
- https://github.com/duncanrhamill/oakd_kimera

And to the question of mapping RGB images, we have an example of that, doing RGB point-cloud projection is here: https://github.com/luxonis/depthai-experiments/tree/master/pcl-projection-rgb

Thoughts?

Oh and there's active discussion/development in the `SLAM` channel on our Discord:
https://discord.gg/EPsZHkg9Nx

Thanks,
Brandon

taloot, 2023-01-31:
> Great question @Zumbalamambo and sorry about the delay. So actually many users are now working on integrating OAK/depthai with SLAM solutions.
>
> Below is a summary of some of the current in-flight [...]

ur answer  is not related to the question with... with oak-d u CANT 3d scan an item what so ever... 
NEVER buy oak-d if you plan to make 3d scanner

Zumbalamambo, 2023-01-31:
and it has been more than 2 years!

taloot, 2023-01-31:
> and it has been more than 2 years!

haha yeah and oak-d should mention that u cant use it as 3d scanner, as they give u impression that u can,but they cant sell it 300 usd, as there is 30-40 usd gadgets, doing the same functions as oakd