luxonis/depthai #324: Pip wheel for Opencv-python doesn't work on PiZero

eric-schleicher, 2021-01-27:
Create to separate from similar but unrelated issue for jetson platform from issue #309 

**Describe the bug**
pipi wheel for opencv-python doesn't seem to work on pi zero

**To Reproduce**
Steps to reproduce the behavior:
1. `python3 -m pip install opencv-python`
2. `python3 -c "import cv2; import numpy as np; blank_image = np.zeros((500,500,3), np.uint8); cv2.imshow('s', blank_image); cv2.waitKey(0)"` 

**Expected behavior**
get black box.

**Screenshots**
//TBA

Luxonis-Brandon, 2021-01-27:
I can replicate this on a Pi Zero:
[image]
The -exact- same SD-Card on a Pi 3B+ does not have the issue, though:
[image]

So this is definitely something specific to the Pi Zero.  We are investigating what is happening here.

Luxonis-Brandon, 2021-01-28:
Same issue here: https://discuss.luxonis.com/d/91-oak-d-on-raspberry-pi-zero-version-not-satisfied-open3d-0-10-0-0/9

closed 2021-02-05 by szabi-luxonis

jremington, 2023-06-26:
Hi, Folks:
I tried following these steps for the Oak-D Lite with a new Pi Zero W and along the line it seems that Python 3.10 was required. So I installed that successfully, then discovered that no wheel has been built for this architecture and Python 3.10.

Attempting to build that led rather quickly to a failure with message that means nothing to me (quoted below). Suggestions for proceeding, including for downgrading from Python 3.10, would be appreciated!

`pi@pizero:~/Luxonis/depthai $ python3 install_requirements.py
pip 23.1.2 from /usr/lib/python3.10/site-packages/pip (python 3.10)
[WARNING] There are no prebuilt wheels for Python 3.10 for OpenCV, building process on this device may be long and unstable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/lib/python3.10/site-packages (23.1.2)
WARNING: Skipping opencv-python as it is not installed.
WARNING: Skipping opencv-contrib-python as it is not installed.
WARNING: Skipping depthai as it is not installed.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring pyqt5: markers 'platform_machine != "armv6l" and platform_machine != "armv7l" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
Collecting requests==2.26.0 (from -r requirements.txt (line 1))
  Using cached https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)
Collecting numpy>=1.21.4 (from -r requirements.txt (line 3))
  Using cached numpy-1.25.0.tar.gz (10.4 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting opencv-contrib-python==4.5.5.62 (from -r requirements.txt (line 4))
Traceback (most recent call last):
  File "/home/pi/Luxonis/depthai/install_requirements.py", line 45, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"], cwd=script_directory)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '--user', '-r', 'requirements.txt']' died with <Signals.SIGKILL: 9>.`

jremington, 2023-06-26:
Hi, Folks:

I have the Oak-D Lite and would like to have it running with the Pi Zero W. I followed the discussion at https://discuss.luxonis.com/d/91-oak-d-on-raspberry-pi-zero-version-not-satisfied-open3d-0-10-0-0/16

But in the meantime it appears the Python3.10 is required, and no wheel has been built for that and the Pi Zero.
I've installed Python 3.10, but and then try to build the wheel, which fails quickly with the following error messages.  Suggestions on how to proceed would be most appreciated!

`pi@pizero:/usr/lib/python3.10 $ echo $PATH
/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/home/pi/Luxonis/depthai/entrypoint
pi@pizero:/usr/lib/python3.10 $ cd ~/Luxonis/depthai
pi@pizero:~/Luxonis/depthai $ ls
apps                   depthai_demo.py           entrypoint               log_system_information.py
autostart              depthai_helpers           epipolar_test_online.py  README.md
calibrate_and_test.py  depthai_sdk               gui                      requirements-optional.txt
calibrate.py           docker_dependencies.sh    install_requirements.py  requirements.txt
callbacks.py           Dockerfile                launcher                 resources
charuco_11x8.pdf       dynamic_recalibration.py  LICENSE                  tests
pi@pizero:~/Luxonis/depthai $ python3 install_requirements.py
pip 23.1.2 from /usr/lib/python3.10/site-packages/pip (python 3.10)
[WARNING] There are no prebuilt wheels for Python 3.10 for OpenCV, building process on this device may be long and unstable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /usr/lib/python3.10/site-packages (23.1.2)
WARNING: Skipping opencv-python as it is not installed.
WARNING: Skipping opencv-contrib-python as it is not installed.
WARNING: Skipping depthai as it is not installed.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring pyqt5: markers 'platform_machine != "armv6l" and platform_machine != "armv7l" and platform_machine != "aarch64" and platform_machine != "arm64"' don't match your environment
Collecting requests==2.26.0 (from -r requirements.txt (line 1))
  Using cached https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)
Collecting numpy>=1.21.4 (from -r requirements.txt (line 3))
  Using cached numpy-1.25.0.tar.gz (10.4 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Collecting opencv-contrib-python==4.5.5.62 (from -r requirements.txt (line 4))
Traceback (most recent call last):
  File "/home/pi/Luxonis/depthai/install_requirements.py", line 45, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"], cwd=script_directory)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '--user', '-r', 'requirements.txt']' died with <Signals.SIGKILL: 9>.
pi@pizero:~/Luxonis/depthai $`

PR #329 (2021-02-05): Add custom wheel for armv6 (pi zero) + aarch64 (jetson)
The wheel is built on my pi zero, and downloaded from google drive.
Tested on clean Raspberry OS (latest) following the docs (same as any other raspberry)
`sudo curl -fL http://docs.luxonis.com/_static/install_dependencies.sh | bash`
`git clone https://github.com/luxonis/depthai.git`
`python3 install_requirements.py`

`python3 depthai_demo.py -rgbf 5`

Fixes #324