luxonis/depthai #326: Example Request of Automatically Storing JPEGs for a Detected Class

Luxonis-Brandon, 2021-02-01:
### Start with the `why`:

As more users are getting up to speed with depthai, one of the most commonly requested features is automatically saving a JPEG when a certain class is detected.  Such a thing is write-able by the end user, but we have not given advice in many locations for the same exact thing.

### Move to the `how`

We should write a canonical example of doing this, using the Gen2 pipeline builder, and add it to the code samples here: [https://docs.luxonis.com/projects/api/en/gen2_develop/](https://docs.luxonis.com/projects/api/en/gen2_develop/)

### Move to the `what`:

Make an example that automatically saves a JPEG when a certain class is detected, e.g. `person` or `tv/monitor` for the default MobileNetSSDv2 PASCAL VOC 2007-trained model.

VanDavv, 2021-02-04:
@Luxonis-Brandon here is the initial implementation https://github.com/luxonis/depthai-experiments/pull/62

qaqak, 2021-02-05:
> @Luxonis-Brandon here is the initial implementation [luxonis/depthai-experiments#62](https://github.com/luxonis/depthai-experiments/pull/62)

What about the other models?

qaqak, 2021-02-15:
luxonis/depthai-experiments#62
This is not what I meant.
I meant that you can run a Python script or function whenever something is detected.

Luxonis-Brandon, 2021-02-28:
https://github.com/luxonis/depthai-experiments/tree/master/gen2-class-saver-jpeg

Luxonis-Brandon, 2021-02-28:
> > @Luxonis-Brandon here is the initial implementation [luxonis/depthai-experiments#62](https://github.com/luxonis/depthai-experiments/pull/62)
> 
> What about the other models?

Is there a specific model you are wanting to see?

closed 2021-02-28 by Luxonis-Brandon

qaqak, 2021-02-28:
Yes
`vehicle-license-plate-detection-barrier-0106`

Luxonis-Brandon, 2021-03-02:
So are you wanting to save when a vehicle is detected?  If so actually it's just a matter of changing the model and then the class to save.  We can write up an example if you'd like.  Thoughts?

qaqak, 2021-03-02:
Yes, thank you.
And I am also trying run a function when a vehicle is detected.
If you want, you can see my current code

wasauce, 2021-03-02:
@qaqak while I'm not part of the Luxonios team -- post what code you have and I'll happily take a look

Luxonis-Brandon, 2021-03-02:
Thanks @wasauce .  And yes we'd like to see it as well.  CC: @cafemoloko .

qaqak, 2021-03-03:
> @qaqak while I'm not part of the Luxonios team -- post what code you have and I'll happily take a look

I put it in [this repo](https://github.com/qaqak/license-plate-recognition-with-opencv) with a README.md

dhruvsheth-ai, 2021-03-04:
Hey @qaqak , I see that you are using Gen1 API to use the pre-trained OpenVINO Model. The Gen2 API is comparitively easy to customize and there are documented scripts you could follow along. Would you mind giving the Gen2 API a shot with this model. You could convert the .xml and .bin files from the OpenVINO model `vehicle-license-plate-detection-barrier-0106
` and convert to .blob file using the detailed doc page - https://docs.luxonis.com/projects/api/en/latest/tutorials/local_convert_openvino/

The OpenVINO Model is a Mobilenet V2 SSD, and fits the implementation in the gen2 class saver jpeg example - https://github.com/luxonis/depthai-experiments/tree/master/gen2-class-saver-jpeg

Also edit the labels to the one's the model would be using to ensure there are no errors
This would leave you with more room to flexibility and space for adding custom functions.