luxonis/depthai #332: [BUG]

LaurentBerger, 2021-02-12:
**Check if issue already exists**
I don't follow documentation : i don't use virtualenv on windows10

**Describe the bug**
I run run_test.py 
```
G:\Lib\depthai>run_tests.py
No calibration file. Using Calibration Defaults.
'F:\Program' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
Test meta_d2h,metaout: Failed!
Test meta_d2h,metaout: Rerun...
'F:\Program' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
Test meta_d2h,metaout: Failed!
Traceback (most recent call last):
  File "G:\Lib\depthai\tests\tests_runner.py", line 55, in <module>
    runner.test_streams("meta_d2h metaout")
  File "G:\Lib\depthai\tests\tests_runner.py", line 42, in test_streams
    self.test(name, cmd)
  File "G:\Lib\depthai\tests\tests_runner.py", line 35, in test
    self.test(name, cmd, run_nr=run_nr + 1)
  File "G:\Lib\depthai\tests\tests_runner.py", line 37, in test
    raise RuntimeError("Test failed: " + cmd)
```
RuntimeError: Test failed: F:\Program Files\Python\python.exe G:\Lib\depthai\tests\test_executor.py -s meta_d2h metaout

Now If change [source code ](https://github.com/luxonis/depthai/blob/main/tests/tests_runner.py#L39-46)
in 

```
    def test_streams(self, streams):
        cmd = '"' + sys.executable + '"' + " " + rel_path('./test_executor.py') + ' -s ' + streams
        name = ','.join(streams.split(' '))
        self.test(name, cmd)

    def test_cnn(self, cnn):
        cmd = '"' + sys.executable + '"' + " " + rel_path('./test_executor.py') + ' -s previewout metaout -cnn ' + cnn
        self.test(cnn, cmd)

```
full output with new code is here
[oak_d_run_tests.txt](https://github.com/luxonis/depthai/files/5972138/oak_d_run_tests.txt)

Luxonis-Brandon, 2021-02-12:
Thank you for the report @LaurentBerger .  The run_tests.py could very well be outdated.  @VanDavv - could you check on this when you have a second?

VanDavv, 2021-02-23:
I just gave it a shot on my Linux machine and it all works well

```
Test human-pose-estimation-0001: Passed!
Test meta_d2h,metaout: Passed!
Test previewout: Passed!
Test left: Passed!
Test right: Passed!
Test depth: Passed!
Test disparity: Passed!
Test disparity_color: Passed!
Test object_tracker: Passed!
Test age-gender-recognition-retail-0013: Passed!
Test person-vehicle-bike-detection-crossroad-1016: Passed!
Test vehicle-license-plate-detection-barrier-0106: Passed!
Test face-detection-adas-0001: Passed!
Test mobilenet-ssd: Passed!
Test yolo-v3: Passed!
Test person-detection-retail-0013: Passed!
Test deeplabv3p_person: Passed!
Test face-detection-retail-0004: Passed!
Test openpose: Passed!
Test openpose2: Passed!
Test mobileNetV2-PoseEstimation: Passed!
Test pedestrian-detection-adas-0002: Passed!
Test tiny-yolo-v3: Passed!
Test vehicle-detection-adas-0002: Passed!
Test human-pose-estimation-0001: Passed!
```

What I think that might be a case is this log line

> 'F:\Program' n’est pas reconnu en tant que commande interne
> ou externe, un programme exécutable ou un fichier de commanders.

Which after translation, means

> 'F:\Program' is not recognized as an internal command
> or external, an executable program or a batch file.

So I'll give it a try also on my Windows system

VanDavv, 2021-02-23:
Ok, so I tested it on windows and the tests works there too. 
@LaurentBerger could you try again as soon as [this pr](https://github.com/luxonis/depthai/pull/338) gets merged?

LaurentBerger, 2021-02-25:
@vanDavy I tried PR and result is the same
```
git pull
git checkout develop
G:\Lib\depthai>run_tests.py
No calibration file. Using Calibration Defaults.
'F:\Program' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
Test meta_d2h,metaout: Failed!
Test meta_d2h,metaout: Rerun...
'F:\Program' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.
Test meta_d2h,metaout: Failed!
Traceback (most recent call last):
  File "G:\Lib\depthai\tests\tests_runner.py", line 56, in <module>
    runner.test_streams("meta_d2h metaout")
  File "G:\Lib\depthai\tests\tests_runner.py", line 43, in test_streams
    self.test(name, cmd)
  File "G:\Lib\depthai\tests\tests_runner.py", line 36, in test
    self.test(name, cmd, run_nr=run_nr + 1)
  File "G:\Lib\depthai\tests\tests_runner.py", line 38, in test
    raise RuntimeError("Test failed: " + cmd)
RuntimeError: Test failed: F:\Program Files\Python\python.exe G:\Lib\depthai\tests\test_executor.py -s meta_d2h metaout
```
In windows quotation marks ( symbol ") must be used  in command line when there is a space un folder name. Example in windows command: 
```
G:\Lib\depthai>F:\Program Files\Python\python.exe
'F:\Program' n’est pas reconnu en tant que commande interne
ou externe, un programme exécutable ou un fichier de commandes.

G:\Lib\depthai>"F:\Program Files\Python\python.exe"
Python 3.8.5 (tags/v3.8.5:580fbb0, Jul 20 2020, 15:57:54) [MSC v.1924 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
```
https://stackoverflow.com/questions/6376113/how-do-i-use-spaces-in-the-command-prompt

dhruvsheth-ai, 2021-03-10:
Your F: drive may be an external storage device, such as a flash drive. The F: drive on your computer refers to a secondary storage device, so does not have executing permissions. Maybe, you could change the path and try from `C:\`? 
Let me know if that helps