luxonis/depthai #333: [BUG] Udev check in config_manager fails if udev rules with bad permissions exist in /etc/udev/rules.d

botmayank, 2021-02-16:
**Describe the bug**
I had STlink related rules lying around in my `/etc/udev/rules.d` with permissions preventing the `grep` command (linuxCheckApplyUsbRules() in config_manager.py) from running on them. All other good rules had octal permission 644 while these had 700. This caused `python3 depthai_demo.py` to fail and complain about udev rules even though I had properly added the OAK-D ones.

**To Reproduce**
Steps to reproduce the behavior:
1. Add a rule with file permissions 700. `sudo chmod 700 49-stlinkv1.rules`
2. Try launching the demo app: `python3 depthai_demo.py`
3. It'll fail with the log for udev rules:
```
Depthai version installed:  0.4.1.1

WARNING: Usb rules not found

Set rules: 
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules 
sudo udevadm control --reload-rules && udevadm trigger 
Disconnect/connect usb cable on host! 
```

**Expected behavior**
The demo app should've launched.

**SOLUTION**
Simply changing the octal permissions for the offending rules fixed the issue.

`sudo chmod 644 <annoying.rules>`

botmayank, 2021-02-16:
Described fix in issue description^^

closed 2021-02-16 by botmayank

Luxonis-Brandon, 2021-02-16:
Awesome, thanks for taking the time to write this up @botmayank so that others will know how to fix this when running into the same or similar issue!