luxonis/depthai #335: Object tracker sample code not working

jonathanleinola, 2021-02-17:
Object tracker sample: https://docs.luxonis.com/en/latest/pages/samples/object_tracker/ is not working.

In the pipeline creation the blob file seems to point out to resource which does not exist:

p = device.create_pipeline(config={
    "streams": ["previewout", "object_tracker"],
    "ai": {
        #blob compiled for maximum 12 shaves
        "blob_file": "/path/to/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.blob.sh12cmx12NCE1",
        "blob_file_config": "/path/to/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json",
        "shaves" : 12,
        "cmx_slices" : 12,
        "NN_engines" : 1,
    },
    'ot': {
        'max_tracklets': 20,
        'confidence_threshold': 0.5,
    },
})

Also if the file is pointed to for example:
    "ai": {
        #blob compiled for maximum 12 shaves
        "blob_file": "/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.blob.sh14cmx14NCE1",
        "blob_file_config": "/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json",
        "shaves" : 14,
        "cmx_slices" : 14,
        "NN_engines" : 1,
    },
    'ot': {
        'max_tracklets': 20,
        'confidence_threshold': 0.5,
    },
})

Then i get this error: ERROR: requested CNN resources overlaps with RGB camera :( Any help?

Luxonis-Brandon, 2021-02-17:
Hi @jonathanleinola ,

Sorry about the trouble here.  I had been trying to figure out what is going wrong but didn't notice at first.  So I think you need to also specify `metaout` in your streams.  I think that is what is missing.  

I think this is it, as if I specify `metaout` along with `previewout` and `object_tracker` in `depthai_demo.py` it does succesfull track my `boat` (the ceiling of the garage from which I am working):

[image]

This command was used:
`depthai_demo.py -s metaout previewout object_tracker`

But if I run without requesting `metaout` then there is no neural network data, and so no tracker (as the tracker works off the results of the neural network object detector output):

[image]

This was run with:

`python3 depthai_demo.py -s previewout object_tracker `

And notice no `boat` and no tracking/etc. and the NN FPS shows as 0.  In other words, not requesting the metaout from the neural network tells the firmware to not run the neural network.

So I think the only thing you need to change (keeping the sh12cmx12 model, your first attempt) is to add `"metaout"` to the streams, as below:

```
p = device.create_pipeline(config={
"streams": ["metaout", "previewout", "object_tracker"],
...
```
So the `python3 depthai_demo.py -s metaout previewout object_tracker` command is the CLI-equivalent of that.

Thoughts?

Thanks,
Brandon

jonathanleinola, 2021-02-18:
Thanks! I think i'm now one step closer, but still i'm missing this file: mobilenet-ssd.blob.sh12cmx12NCE1? The script can not find it / (me neither)

szabi-luxonis, 2021-02-18:
> Thanks! I think i'm now one step closer, but still i'm missing this file: mobilenet-ssd.blob.sh12cmx12NCE1? The script can not find it / (me neither)

Run `python3 depthai_demo.py -sh 12`, it will generate the required model : `mobilenet-ssd.blob.sh12cmx12NCE1` at `depthai/resources/nn/mobilenet-ssd/`
Alternativel if you would follow the steps Brandon described previously, it would work as it is.
`python3 depthai_demo.py -s metaout previewout object_tracker`

jonathanleinola, 2021-02-18:
Oh, thanks!! It's working now :)

jonathanleinola, 2021-02-18:
Any ideas on how could I get the spatial data from the tracklet object? (x,y,z position)

Luxonis-Brandon, 2021-02-18:
Hi @jonathanleinola ,

So unfortunately we do not have that functionality implemented yet.  We had intended to have it out in December but are behind.  We are planning on implementing it in our Gen2 API (examples [here](https://docs.luxonis.com/projects/api/en/gen2_develop/)), and the issue for it is https://github.com/luxonis/depthai/issues/293.

Largely we've ended up spending a lot more time then we expected on `unknown unknowns` discovered in functionality we already wrote in Gen2 and then fixing/refactoring/etc. to make this work properly/better.  Fortunately, a good number of these actually do involve making it so that tracking in 3D coordinates should work better and should be easier to implement.  

I'm not sure on timeline for it yet, but @szabi-luxonis has been doing a lot of underlying work recently to allow this.  So he might have an idea as to when we will have this out in Gen2.  My gut is we'll probably have it in March.

In the meantime, it is actually possible to use Gen2 to grab the depth region of interest from a detected object or any ROI.  So you could actually use something like mosse (e.g. [here](https://www.pyimagesearch.com/2018/07/30/opencv-object-tracking/) run on the host, and our depth-averager node to accomplish 3D position of a tracked object.

So the flow would be: take ROI from object detection in Gen2, pass this to mosse iteratively, use the mosse ROI and feed this into depthai to grab the depth from the ROI, and connect this to the depth averager (another node) to get the 3D position.  (Alternatively you could just have the depth of the ROI returned to the host so you could do more sophisticated processing on it).

I know the depth averager node works (as we have several customers using it over SPI interface), but I don't know if we have an example yet in Python.  So I'll ask and circle back.

Thoughts?

Thanks,
Brandon

jonathanleinola, 2021-03-02:
Hi Brandon,
What is the current situation with the development? Is there some experimental branch on the spatial data from the tracklet object? Also, thanks for the tips, I'll need to do some research on those.

Luxonis-Brandon, 2021-03-02:
Hi @jonathanleinola ,

So the unfortunate news is that we had intended to have this implemented in December, but just ended up behind.  So what is blocking it in Gen2 right now is a crash that occurs with our depth-averager node.  This is what would be used with tracking to give the XYZ data of a tracked object.  Once that is working (not sure on ETA, but @szabi-luxonis is actively on it), we will be able to shift to porting the tracker implementation from the Gen1 API to Gen2 and then integrating it with this depth-averager node.

Not sure on ETA... thinking this month.  

In the meantime, we could probably make an example that runs the tracker and the averaging on the host (which would still run decently actually, even on a Pi, probably still 30FPS).  Would that be helpful?  We'd likely use mosse and the crop-node on device to get the ROI of the mosse tracklet, and then use averaging on the host to get the depth and then XYZ.

Thoughts?

Thanks,
Brandon

jonathanleinola, 2021-03-03:
Hi @Luxonis-Brandon,
Yes, that would be really helpful and good luck with the development!

-Jonathan

Luxonis-Brandon, 2021-03-04:
Thanks and will do!

austy81, 2021-03-19:
Hi, I'm also looking for Object tracking to work (ideally together with depth perception).
Currently I'm adopting [CentroidTracker](https://github.com/luxonis/depthai-experiments/blob/master/people-tracker/modules.py) to work with Gen2 object detection.
But best solution would be to combine object detection and tracking in single NN. Is this also in plan?

Luxonis-Brandon, 2021-03-19:
Thanks @austy81 .  So actually @szabi-luxonis is working now on integrating an improved object tracker as a node in the Gen2 API.

So this would work separately with the NN (so that any NN that produces bounding boxes could be used with it).  I'm not sure how far that work is yet, but I suspect we'll probably at least have a PR with this functionality in the coming weeks.

And on an NN that has tracking built-in, yes, I think that's a great solution.  And such NNs should theoretically already be runnable, but we haven't yet tried them.  

I know AlexeyAB has done a YOLO-LSTM implementation that does this.  Note sure if PINTO0309 has any of the object detection + tracking networks already converted.  

https://github.com/PINTO0309/PINTO_model_zoo

Thanks,
Brandon