luxonis/depthai #336: double speed with out-of-sync interleaved cameras

gukii, 2021-02-18:
Interleave the two cameras out-of-sync exactly, instead of in-sync. This way you can theoretically double the speed from 118 fps to 232 fps.

This is a feature request.

Luxonis-Brandon, 2021-02-18:
Thanks for writing up the feature request.

Luxonis-Brandon, 2021-02-18:
Oh this is also here: https://github.com/luxonis/depthai/issues/175

gukii, 2021-02-18:
Great. Thanks. Must have missed it when scrolling thru the issues.

Luxonis-Brandon, 2021-02-18:
Thanks.  So this looks quite doable from the first look.  We just need to crank out some critical items (like depth alignment, IMU support, object tracker in Gen2, etc.) and then will be able to implement this.  So I'm thinking likely April-ish for this.

eric-schleicher, 2021-02-19:
@gukii I'm curious, what kind of application/use-case benefits from this arrangement of alternating frames from different imager positions?

gukii, 2021-02-20:
@eric-schleicher, this is used for measuring / tracking eye-blink speed (how fast the eye lid closes and opens, not just its frequencey) and tracking eye-saccades. achieving higher frame rates is more important than the different imager positions. In a sophisticated implementation, the different positions could improve accuracy (e.g. work around reflections on glasses-wearing users).