luxonis/depthai #337: Support for USB Frame Input

billythedummy, 2021-02-20:
Hello,

We currently have a use case for the BW1099 where we need to process live RGB camera data in another Linux SOM (like a Raspberry Pi) first before running some NN inference on the frames using the BW1099 module. The BW1099 is connected to the Linux SOM via USB.  We were just wondering if there's library support for this specific use case  or if the best approach is to just use OpenVINO and treat the BW1099 as a USB Neural Compute Stick. Thank you!

alex-luxonis, 2021-02-20:
Yes, with Gen2 API it's possible to feed frames from host instead of camera. Here we have an example:
https://docs.luxonis.com/projects/api/en/gen2_develop/samples/17_video_mobilenet/

closed 2021-02-20 by billythedummy

billythedummy, 2021-02-24:
The example is using the host's opencv to crop the frame to the required (300, 300). Is there a way to just feed in the raw full resolution frame and let the module do the cropping? I can't seem to find any example that does it but from the API something like this should just work right?

```
xin = pipeline.createXLinkIn()
im = pipeline.createImageManip()
nn = pipeline.createNeuralNetwork()

im.setResize(300, 300)

xin.out.link(im.inputImage)
im.out.link(nn.input)
```

Luxonis-Brandon, 2021-02-24:
Yes, I think so.  I think that the ImageManip node can be used for this.  See here: [https://docs.luxonis.com/projects/api/en/gen2_develop/references/python/?dummy=http://#ImageManip](https://docs.luxonis.com/projects/api/en/gen2_develop/references/python/?dummy=http://#ImageManip).

Also this may be relevant to making this work: [https://docs.luxonis.com/projects/api/en/gen2_develop/samples/14_color_camera_control/](https://docs.luxonis.com/projects/api/en/gen2_develop/samples/14_color_camera_control/)

And sorry I'm not fully in the code otherwise I'd give more concrete examples.  I'll ask the team as well.

Thoughts?

Thanks,
Brandon

Davidsastresas, 2021-12-21:
Hello Brandon,

I ended up in this thread looking for information for an unrelated project, but I would really love to take a look at the links you provided above, but it asks for a particular user or something like that. Should I be able to see those?

Thanks!

Luxonis-Brandon, 2021-12-21:
Hi @Davidsastresas ,

Great question.  I just fixed the links above.  Sorry about the broken/old/dead links there previously.

You should now be able to see them.

Thanks,
Brandon