luxonis/depthai #341: Openpose + depth  not working togeter in depthai demo

alberdibu, 2021-03-03:
Hi all! Thanks for this great work and the incredible capabilities of the camera. I'm trying to run openpose model and depth estimation to get the pointcloud, both at the same time but if depth and rectified_right streams are available, cnn model doesn't run due to a concrete variable setted to false. It is possible run both methods at the same time. Thanks in advance!!

closed 2021-03-03 by alberdibu

Luxonis-Brandon, 2021-03-03:
What this already resolved @alberdibu ?  Just noticed the closing before I got to it.

alberdibu, 2021-03-03:
Sorry i have seen at least in the code that depth is not available when you run cnn with raw output like in openpose

Luxonis-Brandon, 2021-03-04:
Got it thanks.  So we'll make an example for this in Gen2.  I can't recall if it's already on the action item list.  @VanDavv do you know?

VanDavv, 2021-03-13:
It's already added @Luxonis-Brandon

Luxonis-Brandon, 2021-03-16:
So @alberdibu - if you run just standard depth (not extended disparity, subpixel, or LR-check).  That said, we will produce an example showing how to do this.  

We could actually use the new deptha-averaging node for this @szabi-luxonis , using it for say 7x7 pixel area around each pose keypoint.

Shihnaz, 2021-06-21:
@VanDavv  @Luxonis-Brandon   hello  where  I can find Openpose  with  depth example ,since Openpose + depth  are not working together , please reply ASAP

VanDavv, 2021-06-23:
@Shihnaz could you try running `depthai_demo.py -cnn openpose2`? I just gave it a try and I can see both the nn results and depth output