luxonis/depthai #343: OAK-D not recognized after Myriad X compilation command

luiscastro1995, 2021-03-11:
**Describe the bug**
I was trying to compile yolov5_s into .blob format, which I succeeded, see log example:
```bash
$MYRIAD_COMPILE -m yolov5s_opt.xml -ip U8 -VPU_NUMBER_OF_SHAVES 4 -VPU_NUMBER_OF_CMX_SLICES 4
Inference Engine: 
	API version ............ 2.1
	Build .................. 2021.2.0-1877-176bdf51370-releases/2021/2
	Description ....... API
[Warning][VPU][Config] Deprecated option was used : VPU_MYRIAD_PLATFORM
Done
```
I did this compilation several times, because I was getting errors when using the generated blob file for inference on the OAK-D. The compilations were done with the OAK-D unplugged, except for one time where I compiled for Myriad X with the device connected. That time the Log was different (I am sorry but I did not save the log to show you), there was an error message  before the "Done". The command that generated the strange log was 
```bash
$MYRIAD_COMPILE -m yolov5s_opt.xml -ip FP32 -op FP32
```
Unfortunately, this log does not appear anymore, even if I run the same command.

Now comes the strange part. After that point I am not able to run any model on the device even the examples. I tried to run the examples from Gen1 and Gen2 API. For the Gen1, it waits 10 seconds and it does not find any device. For the Gen2 examples it outputs the following:
```bash
with dai.Device(pipeline) as device:
RuntimeError: No available devices
```
So again no device found.

I tried to see if the camera was being recognized by running on the terminal: **lsusb** with the device connected and not connected: There was no difference on the output:
```bash
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 13d3:3491 IMC Networks 
Bus 001 Device 003: ID 138a:0010 Validity Sensors, Inc. VFS Fingerprint sensor
Bus 001 Device 002: ID 0bda:58b3 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I have tried using other USB port (3 available all 3.0) and using Windows 10, which does not recognize the device as well.

Moreover, I could not replicate the strange log during the compilation for the Myriad X. If I try to run the same command

Any hints?

Luxonis-Brandon, 2021-03-11:
Very odd.  So the fact that it's not showing up at all via USB means that likely the device isn't even powering on enough to enumerate.  

Can you see if the LEDs are turning on when plugging it in (but looking up the USB port area a bit)?

luiscastro1995, 2021-03-11:
I can see a orange light coming out from the usb-c port, so I think power is not a problem.

Is there anyway to "reset" the camera myself?

alex-luxonis, 2021-03-11:
There is no need to reset, and no way actually, as OAK-1 and OAK-D don't have internal storage to boot from, but everything is loaded over USB. A simple power-cycle should fix issues that might appear. 

Is the device powered by the 5V adapter? If yes, could you try removing it and powering over USB only.

Also could you run `dmesg -w` and see if any new activity appears there while plugging and unplugging the USB cable.

luiscastro1995, 2021-03-12:
Removing the power source and only using the usb-c really worked, the device is now recognized, even if I plug in again the power source.
I tried to reproduce again my problem, by plugging the power source and connecting the camera to my computer while doing the MyriafX compilation. Here is the strange Log (line before "Done"):

```bash
luis@luis-P2440UQ:~/Projs_Software/INESC_projects/AOK/YoloV5_Models/Coco_ONNX_Model/openvino/yolov5s_coco$ $MYRIAD_COMPILE -m yolov5s_opt.xml -ip FP16 -VPU_NUMBER_OF_SHAVES 4 -VPU_NUMBER_OF_CMX_SLICES 4
Inference Engine: 
	API version ............ 2.1
	Build .................. 2021.2.0-1877-176bdf51370-releases/2021/2
	Description ....... API
[Warning][VPU][Config] Deprecated option was used : VPU_MYRIAD_PLATFORM
E: [ncAPI] [    191488] [myriad_compile] ncDeviceOpen:1013	Failed to find booted device after boot
Done
```

As before, after this my camera stopped working, unless I unplug it from the power source and only use the usb-c connection.
Really strange but solution found, thank you very much!

alex-luxonis, 2021-03-12:
Ok, so this makes sense, it wasn't clear that the device was not power-cycled.

So what happens is: as our device exposes the MyriadX ROM bootloader interface, OpenVINO loads the NCS2 firmware to it (not exactly clear why it does that, and not sure if there is some option to disable this behavior), but somehow the device fails to boot up with the NCS2 firmware, as it also doesn't show up in `lsusb`. Then it is left in this state, and only a hard-reset (inaccesible) or power-cycle will recover it.

We should check ourselves if this behavior is reproducible with 2021.2, as normally the Luxonis devices should work with OpenVINO directly.

Luxonis-Brandon, 2021-03-12:
This is fascinating.  I hadn't thought of this case.  I'm glad it's not dead!  It would have been our first field-failure.  (We've had a field-destruction, but that is a different type of failure.)