luxonis/depthai #348: [BUG] Windows depthai=2.0.0 is not finding the module

piperod, 2021-03-30:
Once I updated depthai to 2.0  to run the depthai-python examples I run into issues not being able to import the module. 

**To Reproduce**
Steps to reproduce the behavior:
1. pip install depthai==2.0 
2. python -c "import depthai" 
3. error : "module depthai not found" 
This seems to happen only in Windows. In my mac machine seems to be working fine.

Luxonis-Brandon, 2021-03-31:
Thanks for the report.  Will investigate.

Luxonis-Brandon, 2021-03-31:
Hi @piperod ,

So I just heard back.  So 2.0 actually had a bug specific to Windows that made it not install/work on Windows.  This was hotfixed to make 2.0.0.1, so anything 2.0.0.1+ should not have this problem.

Thoughts?

Thanks,
Brandon

piperod, 2021-03-31:
Thanks @Luxonis-Brandon for looking into this.  I will test according to your comments. Thanks.

closed 2021-05-24 by szabi-luxonis