luxonis/depthai #356: [BUG] RuntimeError: Model compiler failed! Not connected to the internet?

marmalodak, 2021-04-17:
What else is needed to run the demo? As far as I can tell, I've installed everything as documented.

I've run this several times to see if it was a transient issue with whatever service is running at host='69.164.214.171', port=8083. I get the same results each time.

    (oak-venv) pi@raspberrypi:~/oak-1/depthai $ ./depthai_demo.py 
    No calibration file. Using Calibration Defaults.
    Using depthai module from:  /home/pi/oak-1/oak-venv/lib/python3.7/site-packages/depthai.cpython-37m-arm-linux-gnueabihf.so
    Depthai version installed:  1.0.0.0
    Compiling model for 14 shaves, 14 cmx_slices and 1 NN_engines 
    model_compilation_target: cloud
    ################|| Downloading models ||################
    
    ========== Retrieving /home/pi/oak-1/depthai/model_compiler/downloads/mobilenet-ssd/FP16/mobilenet-ssd.xml from the cache
    
    ========== Retrieving /home/pi/oak-1/depthai/model_compiler/downloads/mobilenet-ssd/FP16/mobilenet-ssd.bin from the cache
    
    ################|| Post-processing ||################
    
    Unknown error occured: HTTPConnectionPool(host='69.164.214.171', port=8083): Max retries exceeded with url: /compile?version=2020.1 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0xaaf07c50>: Failed to establish a new connection: [Errno 110] Connection timed out'))
    Traceback (most recent call last):
      File "./depthai_demo.py", line 418, in <module>
        dai.startLoop()
      File "./depthai_demo.py", line 47, in startLoop
        configMan = DepthConfigManager(args)
      File "/home/pi/oak-1/depthai/depthai_helpers/config_manager.py", line 26, in __init__
        self.generateJsonConfig()
      File "/home/pi/oak-1/depthai/depthai_helpers/config_manager.py", line 192, in generateJsonConfig
        blobMan = BlobManager(self.args, self.calc_dist_to_bb, shave_nr, cmx_slices, NCE_nr)
      File "/home/pi/oak-1/depthai/depthai_helpers/config_manager.py", line 377, in __init__
        self.blob_file = self.compileBlob(self.args['cnn_model'], self.args['model_compilation_target'])
      File "/home/pi/oak-1/depthai/depthai_helpers/config_manager.py", line 431, in compileBlob
        ret = download_and_compile_NN_model(nn_model, model_zoo_folder, shave_nr_opt, cmx_slices_opt, NCE_nr, outblob_file, model_compilation_target)
      File "/home/pi/oak-1/depthai/model_compiler/model_compiler.py", line 182, in download_and_compile_NN_model
        raise RuntimeError("Model compiler failed! Not connected to the internet?")
    RuntimeError: Model compiler failed! Not connected to the internet?
    
    
    (oak-venv) pi@raspberrypi:~/oak-1/depthai $ pip freeze
    argcomplete==1.12.1
    certifi==2020.12.5
    chardet==3.0.4
    depthai==1.0.0.0
    idna==2.10
    importlib-metadata==2.1.1
    numpy==1.19.5
    opencv-python==4.5.1.48
    pkg-resources==0.0.0
    PyYAML==5.3.1
    requests==2.24.0
    urllib3==1.25.11
    zipp==3.4.1
    (oak-venv) pi@raspberrypi:~/oak-1/depthai $ uname -a
    Linux raspberrypi 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
    (oak-venv) pi@raspberrypi:~/oak-1/depthai $ git show
    commit 96d4417c62d2cbb038cd9c0fd99a89c1e22b32bf (HEAD -> main, origin/main, origin/HEAD)
    Merge: 9dcccdc ec54483
    Author: Sachin Guruswamy <43363595+saching13@users.noreply.github.com>
    Date:   Thu Mar 18 13:38:50 2021 -0700
    
        Merge pull request #345 from luxonis/calib-remove
        
        removed default.calib file

Luxonis-Brandon, 2021-04-17:
So the system automatically downloads the beauty models for you but you could instead manually download the models per the documentation for each example here:

https://docs.luxonis.com/projects/api/en/latest/samples/08_rgb_mobilenet/#setup

marmalodak, 2021-04-21:
Thanks for answering. I'm unclear about what I'm downloading. 

Do I download each example? Or is it the .blob file that I download?

Luxonis-Brandon, 2021-04-21:
So you can download the whole example and run it, or download just the model and use it with other code.  CC: @Erol444 in case you have other advice here as well.

Erol444, 2021-04-21:
Hello @marmalodak ,
it seems that our blobconverter API wasn't working at the time you were trying to access it. We apologize for the inconvenience. Could you try accessing the `69.164.214.171:8083` from the RPI with a web browser? Could be that something is blocking the request...

Just tested it and it seems to work for me:
[image]

marmalodak, 2021-04-21:
Yesterday evening the depthai demo worked for me. I was able to connect a camera and see some things being detected.

I haven't yet had time to figure out how to cache all the necessary files locally.

Thanks for helping.