luxonis/depthai #366: [BUG]Input tensor 'images' (0) exceeds available data range. Data size (270000B), tensor offset (0), size (3145728B) - skipping inference

sujithgunturu, 2021-05-06:
I have trained an yolo large model to detect wheat heads. I have exported ".pt" weights to ".onnx" model. Later, converted onnx to openvino model using model optimizer. (mo_onnx.py). later used blob converter online too l to convert ".xml" and ".bin" file to ".blob". 

the conversion procedure is ".pt" --> "".onnx' --> ".xml" and ".bin" ---> ".blob"

Then replaced my ".blob" file in "depthai_demo.py" and see how my model performs on the ai kit. I get the following errors and warning. 

`1) [SpatialDetectionNetwork(10)] [warning] Input image (300x300) does not match NN (1024x1024) - skipping inference`
 2)  SpatialDetectionNetwork(10)] [error] Input tensor 'images' (0) exceeds available data range. Data size (270000B), tensor offset (0), size (3145728B) - skipping inference

please help.

dhruvsheth-ai, 2021-05-06:
It seems that the model you've trained is too large. According to the terminal output, it says the model uses 1024 x 1024 image size. This model uses 3145728Bytes which is leager than what the OAK-D can handle. Could you try training your YOLO model with 416 x 416 image size and try again? Maybe that should solve the issue.

sujithgunturu, 2021-05-06:
Since I am skeptical about my model, I have tried few models which are already present in the depthai resources folder for example: tiny-yolo, yolo. and they give the same error.

Luxonis-Brandon, 2021-05-06:
Sorry about the trouble.

@sujithgunturu - can you share the error you are seeing with the included models in the resource folder?  

Thanks,
Brandon

sujithgunturu, 2021-05-06:
1) for the model "depthai/resources/nn/yolo-v3"
```
[error] Input tensor 'inputs' (0) exceeds available data range. Data size (270000B), tensor offset (0), size (519168B) - skipping inference
[14442C1051C8C8D200] [303.126] [SpatialDetectionNetwork(10)] [warning] Input image (300x300) does not match NN (416x416) - skipping inference
```

sujithgunturu, 2021-05-06:
> Sorry about the trouble.
> 
> @sujithgunturu - can you share the error you are seeing with the included models in the resource folder?
> 
> Thanks,
> Brandon

sujithgunturu, 2021-05-06:
> It seems that the model you've trained is too large. According to the terminal output, it says the model uses 1024 x 1024 image size. This model uses 3145728Bytes which is leager than what the OAK-D [...]

I have trained my yolo v5 like this long back. 
python train.py --img 1024 --batch 24 --epochs 100 --data wheat.yaml --cfg models/yolov5l.yaml --name wml

please suggest any other changes.

szabi-luxonis, 2021-05-06:
Yolo-v3 requires 416x416x3 input, you are sending a 300x300x3 input. Just like in the error message is stated.
Change the input image size that you connect as input to the network.

sujithgunturu, 2021-05-06:
> Yolo-v3 requires 416x416x3 input, you are sending a 300x300x3 input. Just like in the error message is stated.
> Change the input image size that you connect as input to the network.

So, I was simply changing the blob link in "depthai_demo.py".  Where do I change the input dimensions of the image.

sujithgunturu, 2021-05-15:
@VanDavv I got "frozen_darknet_yolov3_model.blob" by following [this](https://blog.roboflow.com/deploy-luxonis-oak/) example, How can I run on opencv ai kit. 

There is no test.py in the repo.

sujithgunturu, 2021-05-15:
@VanDavv 
**[14442C1051C8C8D200] [38.130] [SpatialDetectionNetwork(6)] [error] ROI x:0.51809233 y:0.5307379 width:0 height:0.0015344024 is not a valid rectangle.** 

This is what I get for every run

sujithgunturu, 2021-05-15:
@VanDavv 
1) I have run python depthai_demp.py -cnn tiny-yolo-v3
2) then replaces the blob with my custom blob 
3) changes class names in .json file. 
But I could not produce any inference with my custom blob can you help me?

VanDavv, 2021-05-18:
@sujithgunturu could you try running the following command?

```
depthai_demp.py -cnn tiny-yolo-v3 --cnn_input_size 300x300
```

sujithgunturu, 2021-05-18:
@VanDavv 
```
[14442C1051C8C8D200] [94.982] [SpatialDetectionNetwork(6)] [error] Input tensor 'inputs' (0) exceeds available data range. Data size (270000B), tensor offset (0), size (519168B) - skipping inference

[14442C1051C8C8D200] [95.015] [SpatialDetectionNetwork(6)] [warning] Input image (300x300) does not match NN (416x416) - skipping inference
```

sujithgunturu, 2021-05-19:
@VanDavv  link to my custom blob, classes: wheat

https://drive.google.com/file/d/1cR2-i67ru4Zzjg9BMEkC0wNmV7G7eCYB/view?usp=sharing

dhruvsheth-ai, 2021-05-20:
Just tested the blob shared by you. So, I didn't encounter the input size error, but since I haven't used YOLOv5 with OAK-D before, I am not aware of the root problem. So, YOLOv5 isn't officially supported by OpenVINO yet, and the architecture **might** not work with the default script. As soon as I try it on the script - `22_1_tiny_yolo_v3_device_side_decoding.py`, it crashes. It isn't an input size error.

It returns - 
```python
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'rgb' (X_LINK_ERROR)'
```

Any particular reason you're using YOLOv5 over YOLOv4-tiny or YOLOv3-tiny?

sujithgunturu, 2021-05-20:
@dhruvsheth-ai 
The BLOB shared by me was created from this tutorial: https://blog.roboflow.com/deploy-luxonis-oak/
which is yolo_v3.

closed 2021-05-27 by sujithgunturu