luxonis/depthai #369: Should check for None

DavidCooperCheyney, 2021-05-10:
https://github.com/luxonis/depthai/blob/b24451f35c86614831d5b1e86adc4d4146f649db/depthai_demo.py#L549

Change to: 
```python
if conf.args.report is not None and if  len(conf.args.report) > 0):
```

https://github.com/luxonis/depthai/blob/b24451f35c86614831d5b1e86adc4d4146f649db/depthai_demo.py#L561

Change to:
```python
log_out = device.getOutputQueue("system_logger", maxSize=30, blocking=False) if conf.args.report is not None and len(conf.args.report) > 0 else None 
```

Luxonis-Brandon, 2021-05-10:
Thanks for the fix/improvement here!  CC: @VanDavv

VanDavv, 2021-05-11:
Thanks @DavidCooperCheyney, I replaced the default value for `report` param to `[]` (and sorry for the trouble!)

PR with fix - #370

closed 2021-05-11 by VanDavv