luxonis/depthai #381: ModuleNotFoundError: No module named 'consts'

bharath5673, 2021-05-22:
plz help .. im trying to run the script **python run_tests.py**

and im facing thses error

```
Traceback (most recent call last):
  File "tests\tests_runner.py", line 4, in <module>
    from consts.resource_paths import nn_resource_path
ModuleNotFoundError: No module named 'consts'
```

im using conda ,, **python env ver =3.8.10**

alex-luxonis, 2021-05-22:
Oh, these scripts were not ported yet to Gen2, as `depthai_demo.py` is now on `main` branch.
`run_tests.py` may work on `gen1_main` branch though. As well as `calibrate.py`.

@VanDavv / @Erol444 - we should print a message and exit in these scripts, as they could confuse users.

bharath5673, 2021-05-22:
no @alex-luxonis again im facing the same on gen1_main too :(

[image]

VanDavv, 2021-05-27:
@bharath5673 could you check if the tests are now working? (PR got merged)

bharath5673, 2021-06-05:
> @bharath5673 could you check if the tests are now working? (PR got merged)

yes its working , ty ))

closed 2021-06-07 by VanDavv