luxonis/depthai #387: [FeatureRequest] Object detection counting

Erol444, 2021-05-27:
### Start with the `why`:
For quick counting of how many objects of the specified type(label) DepthAI detected in the frame

### Move to the `what`:
Add an argument where you could specify which labels to count eg. `--count 0,1` which would provide the number of detections of labels 0 and 1 on the frame. This should be similar to https://github.com/luxonis/depthai-experiments/pull/136

### Move to the `how`:
Trivial, check https://github.com/luxonis/depthai-experiments/pull/136

Luxonis-Brandon, 2021-06-16:
Does https://github.com/luxonis/depthai-experiments/pull/145 fulfill this @Erol444 ?  At first glance I think it does, but not sure.

Erol444, 2021-06-16:
@Luxonis-Brandon  I don't think so. It's actually a fairly simple ticket, might do it today.

Luxonis-Brandon, 2021-06-16:
Ah OK thanks!

closed 2021-06-21 by Erol444