luxonis/depthai #390: [BUG] Color camera out of synch

domef, 2021-06-01:
**Describe the bug**
When synching the 3 cameras using the sequence number, the color camera is out of synch. More precisely

**To Reproduce**
Steps to reproduce the behavior:
1. Run script (e.g. python3 depthai_demo.py -s color) '....'
2. See error

**Expected behavior**
The three cameras should be synched and the delta  shouldn't vary in time.

**Screenshots**
I saved a frame of the phone timer every minute to approximately measure the 

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

alex-luxonis, 2021-06-17:
This behavior is currently expected: the cameras are started in sequence (RGB starting first), and only Left and Right are synchronized. The sequence numbers are incremented independently per each camera.
About the drift over time, even though the FPS is configured as 30 for all cameras by default, there's a small frame-time difference between IMX378 and OV9282 due to different clock settings / sensor config. It's about 1 frame every 80 seconds.

We'll backport soon the Gen1 sync-related changes from these PRs to fix the issue: https://github.com/luxonis/depthai/pull/157 https://github.com/luxonis/depthai/pull/240

For now, the frame timestamps can be used for the sync purpose.

domef, 2021-06-21:
Thank you for the reply, I will try to use the timestamp.

alex-luxonis, 2021-07-15:
@domef Should be fixed now, see here:
https://github.com/luxonis/depthai-experiments/pull/159
https://github.com/luxonis/depthai-python/pull/310

closed 2024-07-16 by SzabolcsGergely