luxonis/depthai #397: openpose2 installing

R-abodyak, 2021-06-10:
hi ,
when I write $ python depthai_demo.py -cnn openpose2
in the command

I get this problem
"exit_code": 1,
"message": "Command failed with exit code 1, command: /opt/intel/openvino/deployment_tools/inference_engine/lib/intel64/myriad_compile -m /tmp/blobconverter/feea082f408c45eeae0d62e9c8360f42/openpose2/FP16/openpose2.xml -o /tmp/blobconverter/feea082f408c45eeae0d62e9c8360f42/openpose2/FP16/openpose2.blob -c /tmp/blobconverter/feea082f408c45eeae0d62e9c8360f42/myriad_compile_config.txt -ip U8",
"stderr": "The support of IR v5 has been removed from the product. Please, convert the original model using the Model Optimizer which comes with this version of the OpenVINO to generate supported IR version.\n",
"stdout": "Inference Engine: \n\tAPI version ............ 2.1\n\tBuild .................. 2021.3.0-2787-60059f2c755-releases/2021/3\n\tDescription ....... API\n"
}
Traceback (most recent call last):
File "C:\Users\HI TECH\Desktop\depthai\depthai_demo.py", line 574, in
nn_manager = NNetManager(
File "C:\Users\HI TECH\Desktop\depthai\depthai_demo.py", line 198, in init
self.blob_path = BlobManager(model_dir=self.model_dir, model_name=self.model_name).compile(conf.args.shaves)
File "C:\Users\HI TECH\Desktop\depthai\depthai_helpers\config_manager.py", line 313, in compile
self.blob_path = blobconverter.compile_blob(
File "C:\Users\HI TECH\AppData\Roaming\Python\Python39\site-packages\blobconverter_init_.py", line 252, in compile_blob
response.raise_for_status()
File "C:\Users\HI TECH\AppData\Roaming\Python\Python39\site-packages\requests\models.py", line 941, in raise_for_status
raise HTTPError(http_error_msg, response=self)
requests.exceptions.HTTPError: 400 Client Error: BAD REQUEST for url: http://luxonis.com:8080/compile?version=2021.3

what should I do ?

Luxonis-Brandon, 2021-06-10:
I think I saw this issue raised elsewhere and then closed.  Are you still experiencing the problem?

R-abodyak, 2021-06-10:
yes . I still stuck

Luxonis-Brandon, 2021-06-10:
Sorry about the trouble.  I think @VanDavv should be able to help when he wakes up tomorrow (he's in Europe, and wrote the converter that seems to not be responding here).

R-abodyak, 2021-06-11:
thank you. Great! waiting for this

R-abodyak, 2021-06-12:
@Luxonis-Brandon  Hi, could you please find out if VanDavv or anyone can help ? thank you

Luxonis-Brandon, 2021-06-12:
Asking.

VanDavv, 2021-06-12:
Oh, sorry @R-abodyak. I've missed that. Will investigate further and circle back

VanDavv, 2021-06-14:
@R-abodyak the model had to be compiled using the older OpenVINO version (2020.4). I added support for a custom version number in #398 and with that PR merged you should be able to start the network. 

It doesn't have a handler migrated yet though, so the results will be present only as a console print at the moment. I will add it, but you can add your own handler file if you'd like to experiment with it - simply create a python script similar to [this one](https://github.com/luxonis/depthai/blob/main/resources/nn/deeplabv3p_person/handler.py) and add a `handler` config like [here](https://github.com/luxonis/depthai/blob/main/resources/nn/deeplabv3p_person/deeplabv3p_person.json#L6) pointing to this file

VanDavv, 2021-06-14:
Actually, I managed to migrate the handler from Gen1 to work with our handler format - added it to the PR (#398). This means that when that PR is merged, `python3 depthai_demo.py -cnn openpose2` will work and display results

R-abodyak, 2021-06-14:
@VanDavv yes it worked!  Thank you ,, Thank you from the bottom of my heart! <3

VanDavv, 2021-06-15:
Happy to help @R-abodyak :heart:

closed 2021-06-15 by VanDavv