luxonis/depthai #404: Why Depth output is not always available in my pipeline?

laorient, 2021-06-20:
I am currently using saved left and right channel videos to feed into the pipeline and show Depth output. However, it doesn't always work. For example, it may not output depth images for quite a long period, then it will work quite well, and may or may not drop some frames in between. If I use the get() call then it will get stuck quite quickly. If using tryGet(), then it will work but the jumpy nature of it is very frustrating and cannot be useful. How can I go about troubleshooting this problem?

Luxonis-Brandon, 2021-06-21:
Sorry about the trouble here.  I'm not immediately sure but I think @csaba-luxonis / @szabi-luxonis and @alex-luxonis had run into similar before.  I'm not sure if they have any ideas on this or if it's some underlying firmware bug.

laorient, 2021-06-22:
Thanks, @Luxonis-Brandon. If someone can give me some pointers on how to debug/troubleshoot it, that will be great. As long as I can tell, the samples work (both python and C++, also depthai-experiments).

Luxonis-Brandon, 2021-06-22:
Thanks @laorient and sorry about the delay.  I don't immediately know how.  I know @csaba-luxonis has done some work here, so let me see if he can comment on this.

csaba-luxonis, 2021-06-23:
hey @laorient , I ran into that same bug a few weeks ago, I had to send the first frame twice due to the syncing mechanism.

Luxonis-Brandon, 2021-06-23:
Thanks, @csaba-luxonis .  Thoughts, @laorient ?

laorient, 2021-06-26:
> hey @laorient , I ran into that same bug a few weeks ago, I had to send the first frame twice due to the syncing mechanism.

Not very helpful. If you check my code, there was a repeat send for both left frame and right frame at the first time. I commented out the right frame one in the code above. But it doesn't make much difference if I send both channels first frame twice, or just left, or just right. @csaba-luxonis @Luxonis-Brandon

szabi-luxonis, 2021-06-26:
> Not very helpful. If you check my code, there was a repeat send for both left frame and right frame at the first time. I commented out the right frame one in the code above. But it doesn't make much [...]

Due to how syncing mechanism works for stereo frames (right frame is received first, then left) you should send the right frame first, then the left, just as done in the example provided.

Move `qInL.send(imgL)` after `qInR.send(imgR)`

laorient, 2021-06-27:
> Due to how syncing mechanism works for stereo frames (right frame is received first, then left) you should send the right frame first, then the left, just as done in the example provided.
>
> Move [...]

Tried it. I would like to say it is improved. But still way too choppy. It goes on for up to hundreds of frames before it got one or several frames properly processed, missing a lot of interim scenes. I can share with you the videos if anyone is interested. @szabi-luxonis @csaba-luxonis @Luxonis-Brandon

Luxonis-Brandon, 2021-06-27:
How does it look coming from the cameras directly?

laorient, 2021-06-27:
> How does it look coming from the cameras directly?

I used one of the sample app (forgot which one) that captures both channels and saves them to files. Checked the length and frame count of both and they are equal. Opened them individually and verified they are valid.

Luxonis-Brandon, 2021-06-27:
I’m not sure if it would help, but we have a fix in the develop branch that I’m hoping will help.

laorient, 2021-06-27:
> I’m not sure if it would help, but we have a fix in the develop branch that I’m hoping will help.

@Luxonis-Brandon, are you referring to depthai-core or depthai-python?

Luxonis-Brandon, 2021-06-29:
Either, they are in lock-step.

laorient, 2021-07-02:
Tried and it didn't seem to help much, if any. The version used is depthai:2.5.0.0.dev0+2bf6934973b241f6b2c8e7d35e6e9d051d89a81b

Luxonis-Brandon, 2021-07-02:
Got it.  Sorry about the trouble.  So I think I'm not yet understanding the use-case.  Are you using DepthAI as a from-host real-time depth processor?

The reason I ask is that DepthAI is intended for depth directly from the onboard cameras.  We support from-host mainly for being able to do QA of pipelines/etc. - but not in a real-time way.  So there is a high probability that the stack will not necessarily perform real-time when from the host.  

So I just wanted to check on that.  And which model are you using again?

laorient, 2021-07-13:
> Got it. Sorry about the trouble. So I think I'm not yet understanding the use-case. Are you using DepthAI as a from-host real-time depth processor?
>
> The reason I ask is that DepthAI is intended for [...]

I am really trying to test my code using recordings instead of going out to test every time, as my use case is car oriented. I don't need it to perform real-time when from the host, just for it to perform somehow consistently.

Currently I kinda get around it by feeding it a pair of short recordings. It will skip the first dozen or so frames but perform the work more or less consistently for the rest of it. The recordings are 8 seconds long.

szabi-luxonis, 2021-07-13:
@laorient Sorry for the delay on this issue. 
You can check out this branch/codebase, it supports recording / replaying frames, it would fit your needs.
As far as I understand it was fairly well tested and used by a customer.

https://github.com/luxonis/depthai-experiments/tree/replay_memory_fix/gen2-replay