luxonis/depthai #414: [BUG] `[system] [critical] Fatal error. Please report to developers. Log: 'Flic' '89'`

NguyenTuan-Dat, 2021-06-29:
**Describe the bug**
I was try load 5 models to OAK-D but program raise this error:
`
[14442C10E11DCAD200] [17.050] [NeuralNetwork(11)] [warning] Network compiled for 4 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[14442C10E11DCAD200] [17.051] [NeuralNetwork(14)] [warning] Network compiled for 4 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
**[14442C10E11DCAD200] [17.106] [system] [critical] Fatal error. Please report to developers. Log: 'Flic' '89'**
[14442C10E11DCAD200] [17.052] [NeuralNetwork(17)] [warning] Network compiled for 4 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[14442C10E11DCAD200] [17.053] [NeuralNetwork(2)] [warning] Network compiled for 4 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[14442C10E11DCAD200] [17.096] [NeuralNetwork(5)] [warning] Network compiled for 4 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance
[14442C10E11DCAD200] [17.097] [NeuralNetwork(8)] [warning] Network compiled for 1 shaves, maximum available 10, compiling for 5 shaves likely will yield in better performance

RuntimeError: Couldn't read data from stream: '__rpc_main' (X_LINK_ERROR)
`
I guessed shaves had non-enough then i'm was try convert 2 models to 1 shave but this error was raised. This error only raise when i was tried add `palm_detection.blob` model. I'm was tried to replace `palm_detection.blob` with 1 shave running model but program raised the same error

**To Reproduce**
This is my code:
`    def create_nn(self, model_path: str, model_name: str, first: bool = False):
        """

        :param model_path: model path
        :param model_name: model abbreviation
        :param first: Is it the first model
        :return:
        """
        # NeuralNetwork
        print(f"Creating {model_path} Neural Network...")
        model_nn = self.pipeline.createNeuralNetwork()
        model_nn.setBlobPath(str(Path(f"{model_path}").resolve().absolute()))
        model_nn.input.setBlocking(False)
        if first and self.camera:
            print("linked cam.preview to model_nn.input")
            # Create face detection resize manip
            detection_manip = self.pipeline.createImageManip()
            detection_manip.setResize(320, 320)

            # Link camera output to detection_manip input
            self.cam.preview.link(detection_manip.inputImage)

            # Link to face detection neural network input
            detection_manip.out.link(model_nn.input)
            # self.cam.preview.link(model_nn.input)

        else:
            model_in = self.pipeline.createXLinkIn()
            model_in.setStreamName(f"{model_name}_in")
            model_in.out.link(model_nn.input)

        model_nn_xout = self.pipeline.createXLinkOut()
        model_nn_xout.setStreamName(f"{model_name}_nn")
        model_nn.out.link(model_nn_xout.input)

    def create_nns(self):

        if self.camera:
            self.create_nn(
                "models/scrfd_detection_openvino_2021.2_4shave.blob",
                "mfd",
                first=self.camera,
            )

        self.create_nn(
            "models/scrfd_detection_openvino_2021.2_4shave.blob",
            "mfd_db",
            first=False,
        )

        self.create_nn(
            "models/mobilenetv2_50_48_83-0001.blob",
            "gm"
        )

        self.create_nn(
            "models/face-recognition-mobilefacenet-arcface_2021.2_4shave.blob",
            "arcface",
        )

        # Create facial landmarks nn
        self.create_nn(
            # "models/facial-landmarks-35-adas-0002.blob",
            "models/mb1_120x120.blob",
            "landmarks"
        )

        self.create_nn(
            "models/palm_detection.blob",
            "palm"
        )
`

szabi-luxonis, 2021-06-29:
@NguyenTuan-Dat are you using latest depthai version (2.5)? This issue was fixed in 2.1 or 2.2.

NguyenTuan-Dat, 2021-06-29:
@szabi-luxonis I'm using 2.0 version. I'll try to using newer version. Thanks you <3

NguyenTuan-Dat, 2021-06-29:
depthai version 2.2 was resolved my issue <3

closed 2021-06-29 by NguyenTuan-Dat

szabi-luxonis, 2021-06-29:
> depthai version 2.2 was resolved my issue <3

Great, thanks.
I suggest updating to latest regularly since we are improving performance and stability constantly.

NguyenTuan-Dat, 2021-06-29:
> > depthai version 2.2 was resolved my issue <3
> 
> Great, thanks.
> I suggest updating to latest regularly since we are improving performance and stability constantly.

Yeah, i'll keep my system uptodate <3