luxonis/depthai #425: [BUG] {The Sign Language Recognition run very slow on my desktop with oak-d PoE}

JimXu1989, 2021-07-21:
**Check if issue already exists**
Google it (e.g. *error xy* github luxonis depthai)
Check [troubleshooting](https://docs.luxonis.com/en/latest/pages/troubleshooting/) in documentation.

**Describe the bug**
Hi, thanks for sharing the Sign Language Recognition project.
I run the hand_tracker_asl.py on my ubuntu desktop, with my oak-d PoE device, but the frame seems very very low,
The frame will be refresh a few seconds a time, how to improve this?
Thanks a lot!

**To Reproduce**
Steps to reproduce the behavior:
1. Run script (e.g. python3 depthai_demo.py -s color) '....'
2. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Attach system log**
 - Provide output of [log_system_information.py](https://github.com/luxonis/depthai/blob/main/log_system_information.py)

**Additional context**
Add any other context about the problem here.

Erol444, 2021-07-21:
Hello @JimXu1989 , is the problem low FPS or the ~1 second delay you are experiencing? Since you are using the POE device, I wonder if your internet connection throughput is the bottleneck for the frames. Are you connected to the LAN via WiFi? When I had weak signal I experienced similar issues as you. Could you check that out?
Thanks, Erik

Luxonis-Brandon, 2021-07-22:
So one thing that may help here is to port the host-side logic of the sign-language example into the scripting node.  That way there isn't the back/forth logic across the network and on device instead.

Thoughts?

Thanks,
Brandon

chrishem, 2021-07-22:
I'm also having a performance issue with the OAK-D-POE device. I've attached the log_system_information.json file (renamed to .json.log).
[log_system_information.json.log](https://github.com/luxonis/depthai/files/6862750/log_system_information.json.log)

VanDavv, 2021-07-29:
Hi @chrishem and @JimXu1989,

I just submitted a PR that improves the DepthAI demo performance - https://github.com/luxonis/depthai/pull/433

The approach used there can be also applied to other examples / experiments we have