luxonis/depthai #427: [Feature-Request] implement support for high frame rate full resolution for OV tracking camera(s)

eric-schleicher, 2021-07-24:
### Start with the `why`:
High speed imaging is a key aspect of capturing fast moving action (ping pong baseball,etc), SLAM and roi tracking.    Hitting/supporting the imager higher performance modes benefits everyone  and enables depthai to address a broader range of applications and generally to be more capable as a platform. It was also listed as a product feature in the Kickstarter materials. 

### Move to the `what`:

High FPS support was partially implemented as part of https://github.com/luxonis/depthai/issues/174

but stopped short of supporting the higher performance modes the imager supports,  specifically 120FPS full resolution (1280x800) and 170 FPS reduced bit depth

The standard tracking cameras (ov9282 and related variants) support 120 at their native resolution, even higher (170fps) at reduced bit depth.  Implementing the modes the sensor supports maximizes the the opportunity to find the right balance of frame rate and resolution 

### Move to the `how`:
There are example configurations of this sensor that might be useful to help inform

asa, 2021-07-24:
I would love to see the feature tracking #146 support be able to track at full 120+fps internally and then only output an image frame to the host at some lower divisor rate eg 12fps for debug or other purposes which would reduce host bandwidth requirements. Getting the low latency/high frame rate tracker feed would be really useful for lowering the overall latency of the VIO estimates.

asa, 2021-07-24:
I believe this is the config @eric-schleicher is referring to above : [OV9281_MIPI_2Lane_RAW10_8b_1280x800_120fps_ext.cfg](https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Config/USB2.0_UC-391_Rev.E/MIPI/OV9281/2Lane/OV9281_MIPI_2Lane_RAW10_8b_1280x800_120fps_ext.cfg)

Luxonis-Brandon, 2021-07-26:
Thanks both @asa and @eric-schleicher .  We're discussing internally on this as well.  Officially added to the roadmap.

SamKowitt, 2021-09-17:
Can we add the "2021" and "Gen2" label to this enhancement? =)

Luxonis-Brandon, 2021-09-17:
I think we're pretty close.  We've done a lot of underlying driver refactoring which should help with this.  We still have more to go there.  But I think w/in a couple months this should be possible.