luxonis/depthai #430: [Feature-Request]  2 stage NN support

Erol444, 2021-07-28:
### Start with the `why`:
So users could use 2nd stage models like `emotions-recognition-retail-0003`

### Move to the `what`:
Add support for `-cnn2`, so you can specify second stage model as well as the first stage model

### Move to the `how`:
2 NN nodes, and use the first stage detection to parse an image before feeding it into the second stage. We could use Script node to decode the 1st stage and to manage imageManip for cropping the initial image.

VanDavv, 2021-07-28:
I like the idea of using scripting node to decode 1st stage network. Thanks for creating this issue!

Luxonis-Brandon, 2021-07-29:
Yes, agreed.  This will help a ton to make this run performantly on devices that have slow connectivity to the host as well (e.g. POE and/or USB2).

Erol444, 2023-03-27:
Supported in updated [DepthAI SDK](https://docs.luxonis.com/projects/sdk/en/latest/):)

closed 2023-03-27 by Erol444