luxonis/depthai #449: [Feature-Request] {Support Mono Global Shutter cameras in all three ports}

asa, 2021-08-12:
### Start with the `why`:
It would be beneficial to have the camera interfaces be modular so that a mono camera could be used in all three ports of the system, or that RGB cameras could be used in all three ports for instance.
### Move to the `what`:
Support Mono cameras in any port, with the ability to run all three pipelines fully, trigger frames syncronously etc. 
### Move to the `how`:
on Discord it was mentioned there needed to be some abstraction of the camera driver code to support this feature to remove the hardcoded notion of Left, Right and RGB for the camera ports.

asa, 2021-09-09:
> Thanks. Will prepare. If you're not already in our [Discord](https://discord.gg/EPsZHkg9Nx), please join and we can share the temporary branch there. @alex-luxonis is likely who will make it.

any updates on getting a branch out for this?

Luxonis-Brandon, 2021-09-09:
Yes actually it just now went live:
https://github.com/luxonis/depthai-python/pull/365

Good timing!

alex-luxonis, 2021-09-09:
@asa You could try installing the depthai from the above PR / branch `multi_cam_support`:
`python3 examples/install_requirements.py`

And then to run for example 3x OV9282:
`python3 tests/cam_test.py -cams rgb,m left,m right,m`

asa, 2021-09-09:
Cosmic timing!  Yes, this works great.  You all are amazing!