luxonis/depthai #453: [Feature-Request] YOLOX Support

atmccarthy, 2021-08-14:
### Start with the `why`:
The tiny variant of [YOLOX](https://github.com/Megvii-BaseDetection/YOLOX) has a number of advantages over yolov4-tiny:

- [Significantly higher AP](https://arxiv.org/pdf/2107.08430.pdf): **33% v.s. 22%**!
- It's Anchorless
- It's slightly smaller, so it should offer higher frame rates.

There is also a nano variant that would be useful in low power applications, or in situations where you want to run multiple models.

### Move to the `what`:
It would be great if there was first class support for YOLOX in the DepthAI Python and C++ APIs, e.g. by adding support for YOLOX to the existing YoloDetectionNetwork pipeline node. YOLOX is anchorless, so the existing device side decoding probably won't work. I tried it out anyway (by following the OpenVINO [instructions](https://github.com/Megvii-BaseDetection/YOLOX/tree/main/demo/OpenVINO) on the YOLOX github page, and then compiling to a blob for the myriad), but i get the following error:

`[14442C10F1C14ED000] [50.022] [system] [critical] Fatal error. Please report to developers. Log: 'PlgDetectionParser' '109'`

Luxonis-Brandon, 2021-08-16:
Thanks for making this the request @atmccarthy .  So actually if you have this converted to run on Myriad X, it is possible to run it in a non-YOLO parsing way such that the metadata results are sent to the host un-changed and then can be parsed there.

So if this is converted already, this technique can be used instead.  I'll see if I can find a reference of doing this.

Luxonis-Brandon, 2021-08-16:
Found one:
https://github.com/luxonis/depthai-experiments/tree/master/gen2-efficientDet

So this example uses host-side decoding of the neural data.  So DepthAI does not need to know anything about the model being run - DepthAI simply spits out the neural outputs to the host.  

So this is a great way to start for getting YOLOX running if you'd like to get it going now as then (1) we don't block you and (2) we'd need to do it anyway before porting YOLOX parsing directly into the Myriad X, as it's a lot easier to debug such parsings on the host first.

Thoughts?

Thanks,
Brandon

atmccarthy, 2021-08-21:
Thanks @Luxonis-Brandon. I used the example you shared and the YOLOX OpenVINO demo code to build a working solution. Please see the code at the end of this comment, and the attached zip containing a blob for the Myriad. There are some major drawbacks with this approach:

1. I'm using the ESP32 / SPI in a low power application. The final layer of the network has shape (1, 3549, 85), which means you need to pull about 600kb of data back to the host for _each_ frame, and then perform non-maximum suppression via the cpu. The ESP32+ SPI combination can't really handle that much data. The impact on FPS is large even over USB with a powerful CPU at the other end. This problem is less significant if you have a smaller number of classes (v.s. 80 in the COCO dataset). The YOLOX paper does [reference](https://arxiv.org/abs/2101.11782) an approach to doing nms in the model itself, but unfortunately i haven't had the time to explore it further.
2. YOLOX has an image preprocessing / normalization step (see the preproc function in the attached code) that is not part of the model itself, and that can't be done with the ImageManip node. This means we need to pull each camera frame back to the host, normalize it, and then send it back to the device. We could probably get rid of this step during training, although i'm not sure how big the impact on AP would be. YOLOX also expects FP16 data, but AFAIK the XLinkIn only accepts int buffers. I had to work around this by spreading the FP16 pixels across two ints. 
3. My pipeline has downstream nodes (e.g. ObjectTracker), so i need to make another round trip once I've decoded the NN outputs on the host.

As for the performance impact, i get ~15fps running the attached code on my 8700k v.s. ~28 for yolo4 running via the YoloDetectionNetwork node. It's going to be completely unworkable on the ESP, so i haven't tried it.

On the positive side, it does seem to be more accurate than yolov4 in practice, e.g. if i point the camera at my face, then yolo4 says i'm a dog (i had headphones on at the time, which probably didn't help...), v.s YOLOX says i'm a person. YOLOX also detects the picture of my dog that I have sitting on my desk as a dog whereas yolo4 oscillates between cat and dog. I did notice that the YOLOX bounding boxes were a bit janky for really large objects. Anyway, this very scientific test proves that you should look into it further :D

```python
from pathlib import Path
import numpy as np
import cv2
import depthai as dai
import time

def preproc(image, input_size, mean, std, swap=(2, 0, 1)):
    if len(image.shape) == 3:
        padded_img = np.ones((input_size[0], input_size[1], 3)) * 114.0
    else:
        padded_img = np.ones(input_size) * 114.0
    img = np.array(image)
    r = min(input_size[0] / img.shape[0], input_size[1] / img.shape[1])
    resized_img = cv2.resize(
        img,
        (int(img.shape[1] * r), int(img.shape[0] * r)),
        interpolation=cv2.INTER_LINEAR,
    ).astype(np.float32)
    padded_img[: int(img.shape[0] * r), : int(img.shape[1] * r)] = resized_img

    padded_img = padded_img[:, :, ::-1]
    padded_img /= 255.0
    if mean is not None:
        padded_img -= mean
    if std is not None:
        padded_img /= std
    padded_img = padded_img.transpose(swap)
    padded_img = np.ascontiguousarray(padded_img, dtype=np.float16)
    return padded_img, r

def nms(boxes, scores, nms_thr):
    """Single class NMS implemented in Numpy."""
    x1 = boxes[:, 0]
    y1 = boxes[:, 1]
    x2 = boxes[:, 2]
    y2 = boxes[:, 3]

    areas = (x2 - x1 + 1) * (y2 - y1 + 1)
    order = scores.argsort()[::-1]

    keep = []
    while order.size > 0:
        i = order[0]
        keep.append(i)
        xx1 = np.maximum(x1[i], x1[order[1:]])
        yy1 = np.maximum(y1[i], y1[order[1:]])
        xx2 = np.minimum(x2[i], x2[order[1:]])
        yy2 = np.minimum(y2[i], y2[order[1:]])

        w = np.maximum(0.0, xx2 - xx1 + 1)
        h = np.maximum(0.0, yy2 - yy1 + 1)
        inter = w * h
        ovr = inter / (areas[i] + areas[order[1:]] - inter)

        inds = np.where(ovr <= nms_thr)[0]
        order = order[inds + 1]

    return keep

[... 148 lines cut ...]
                final_boxes = dets[:, :4]
                final_scores, final_cls_inds = dets[:, 4], dets[:, 5]

                for i in range(len(final_boxes)):
                    bbox = final_boxes[i]
                    score = final_scores[i]
                    class_name = labelMap[int(final_cls_inds[i])]

                    if score >= 0.1:
                        # Limit the bounding box to 0..SHAPE
                        bbox[bbox > SHAPE] = 1
                        bbox[bbox < 0] = 0
                        xy_min = (int(bbox[0]), int(bbox[1]))
                        xy_max = (int(bbox[2]), int(bbox[3]))
                        # Display detection's BB, label and confidence on the frame
                        cv2.rectangle(frame, xy_min , xy_max, (255, 0, 0), 2)
                        cv2.putText(frame, class_name, (xy_min[0] + 10, xy_min[1] + 20), cv2.FONT_HERSHEY_TRIPLEX, 0.5, 255)
                        cv2.putText(frame, f"{int(score * 100)}%", (xy_min[0] + 10, xy_min[1] + 40), cv2.FONT_HERSHEY_TRIPLEX, 0.5, 255)

        cv2.imshow("rgb", frame)
        if cv2.waitKey(1) == ord('q'):
            break
```

[yolox_tiny.zip](https://github.com/luxonis/depthai/files/7024778/yolox_tiny.zip)

VanDavv, 2021-08-23:
Thanks @atmccarthy for this reference, this is great!
Regarding normalization - we had a discussion about a different network that had the same issue (with nms on model) - you can take a look [here](https://github.com/luxonis/depthai/issues/463#issuecomment-903207334) for the details. Overall, I think this should be handled in the model itself, but for now, I'll check if I can make it work as is add this code to our experiments repo, which can be later optimized.

PINTO0309, 2021-08-23:
Myriad Inference Blob - YOLOX
1. download_tiny.sh
2. download_nano.sh

https://github.com/PINTO0309/PINTO_model_zoo/tree/main/132_YOLOX

This is an old model that was converted within a day after YOLOX was officially released. Please ignore this if it is not helpful as I am not aware of the flow of the discussion.

VanDavv, 2021-08-24:
I gave your snippet a try and I'm a bit uncertain why - for smaller objects, like bottle or person on an image, it works correctly, but for larger ones (like myself in front of the camera), the bounding box is incorrect

[image]

[image]

I added the copied code to [`gen2_yolox`](https://github.com/luxonis/depthai-experiments/tree/gen2_yolox) branch but I think that the model should be modified first (with `--mean_values` and `--scale_values` flags)

atmccarthy, 2021-08-25:
Thanks very much for testing this out @VanDavv. The problem occurs when one of the coordinates of the bounding box is larger than the image. The problem is in this line:

`bbox[bbox > SHAPE] = 1`

It should have been:

`bbox[bbox > SHAPE - 1] = SHAPE - 1`

i.e. the coordinate was being rounded down to 1 instead of 415!

I'll create a PR on your gen2_yolox branch.

atmccarthy, 2021-08-25:
It looks like i don't have permission to push branches to that repo. I've attached a patch to this comment. I made a few other changes:

1. I added a README.md with instructions for creating a blob from a custom YOLOX model + links to the YOLOX repo and this issue.
2. You can now make the pipeline synchronous via a flag (much like you can in the other examples).
3. I renamed some of the variables so that people don't accidentally assume the camera is outputting RGB instead of BGR.

JojoDevel, 2021-12-30:
> The input size has increased with the latest version from 416x416 to 640x640. Also, for some reason, after converting the model to blob, the output is not [1, 3549, 85] but [1, 8400, 85] - although when checking in https://netron.app/, the output layer in ONNX is unchanged

This depends on what kind of yolox model you choose. The `nano` and `tiny` version have `416x416` input res and therefore output a `[1,3549,85]` tensor. All bigger models have `640x640` input res and output a `[1,8400,85]` tensor. So the input size directly correlates with the second dimension of the output size. I assume this is because of their "anchor-free" approach.

All onnx models of yolox can be found in their [repository](https://github.com/Megvii-BaseDetection/YOLOX/tree/main/demo/ONNXRuntime).

> I won't have much time to dig into it in the following days, so will leave it here in case someone else would like to proceed further

I will very soon have a look into YOLOX as soon as my OAK-D Lite device arrives

JojoDevel, 2021-12-30:
@VanDavv @atmccarthy @Luxonis-Brandon Do you know whether it is possible to do the NMS thresholding in hardware like for [YoloV3](https://github.com/luxonis/depthai-core/blob/29e2a5479e9edece375e4596e6561afcc8bfb806/examples/Yolo/tiny_yolo.cpp#L63). I could not find the code for the nms thresholding and, therefore, I don't know the format that it expects ...

tersekmatija, 2021-12-30:
Hey @JojoDevel , 

AFAIK the NMS on device is only done for YoloV3, V4, and V5 (with the 2.14 version). We do not have a direct support for NMS for all the models right now. Since YOLOX is anchorless, you won't be able to mimic the exact outputs of YoloV3,V4 and V5. The Mobilenet node also expects the NN to output a 2D array (n rows x 7 columns) I think, so some part of this post-processing would have to be done on the host any way. If there will be more demand, we'll look into adding native support for YoloX, just like we did with YoloV5 recently :)