luxonis/depthai #477: How to make point cloud with OpenCV OAK-D camera in Python

MrTitusZ, 2021-09-10:
Platform    |     Windows 10
CPU           |     intel i9-9900X
GPU           |     GeForce RTX 2080 Ti
Language  |     Python
Camera     |      OpenCV OAK-D 3D AI Camera Kit
--------------------------------------------

Helo,

I would like to make a point cloud with the OAK-D camera, but i can not find any example code in the GitHub repo.
I have searched on the internet for a solution, but I ony found a complex Rasberry Pi project wich won't run on my computer and I don't really understand that code.
Please could you provide me a simple Python script that can make a point cloud using the OAK-D camera?

Sincerely, TitusZ

MOj0, 2021-09-10:
Hi @MrTitusZ 
We have an example [here](https://github.com/luxonis/depthai-experiments/tree/gen2-rgb-depth-align/gen2-camera-demo), the `main.py` is what you're looking for. In order to run point cloud visualization, use this command: `python main.py -pcl`. We're currently working on a *simpler* version, which will be available next week. I will post the link to it here, once it is done.

MrTitusZ, 2021-09-15:
Helo!

Thank you for that link, it really helps me a lot! :)

I don't know if it is a bug, but when i run python main.py -pcl from cmd, the projection seems wrong.
The point cloud is upside down and the points that are closer to the origin are narrower than those that are further from the origin.

[image]

I did not change anything in the code, maybe I don't understand Open3D well.
And also, how could i make the point cloud colored? (so it uses the RGB sensor to texture the point cloud)

Thank you for your help, I appreciate it!! :)

MOj0, 2021-09-19:
Hi!
That is strange, I get the following result.
[image]

Are you on the master branch?

Anyways, [here](https://github.com/luxonis/depthai-experiments/pull/203) is the PR that I promised. Although there are still changes to be made and the PointCloud is quite noisy, but feel free to play around.