luxonis/depthai #482: [Feature-Request] Add SOT models: ATOM and Dimp18 to model zoo

franva, 2021-09-18:
### Start with the `why`:

SOT (Single Object Tracking) is very helpful in many cases, e.g. a following bot, target tracking etc. The ATOM and DiMP-18 models are SOTA models which won competitions in ICCV 2019 and I have tested all 9 models in the model zoo of [PyTracking](https://github.com/visionml/pytracking/blob/master/MODEL_ZOO.md) and these 2 are relatively quickest and less demanding among other models.

### Move to the `what`:

I would like Depth AI to add the ATOM and DiMP-18 to its model zoo and create a demo code for these 2 models.

### Move to the `how`:

None.

I did notice that the [demos](https://docs.luxonis.com/projects/api/en/latest/samples/object_tracker/?highlight=object%20track) from DepthAI have already have a few object tracking samples, however none of these sample code can perform what the ATOM and DiMP can do, e.g. you can select a particular area on a person(that's right, not necessary the entire body) e.g. left eye, right arm, nose, even a part of your belly, then the 2 models can perform incredible tracking accuracy.

I have found the 2 pre-trained models for your convenience:

1. [ATOM](https://drive.google.com/open?id=1VNyr-Ds0khjM0zaq6lU-xfY74-iWxBvU)
2. [DiMP-18](https://drive.google.com/open?id=1MAjrRJDCbL0DSjUKFyDkUuYS1-cYBNjk)

Hopefully, some talented devs could help to add these 2 models into the DepthAI's ammunition :)

Luxonis-Brandon, 2021-09-21:
Thanks.  Tagging @tersekmatija in case he hasn't seen these already.

franva, 2021-09-21:
hi @ersekmatija , when you are migrating ATOM to the blob format, there are something to avoid, e.g. PrPool
Which is mentioned [here](https://github.com/visionml/pytracking/issues/281#issuecomment-923738982)

tersekmatija, 2021-09-21:
Hi @franva , thanks for sharing this! I agree that such tracking would be better and would have better accuracy than what we have currently available. I'll look into this, but as this will probably require some implementation of additional tracking logic (not just simple forwards through the NN), I am not sure how soon I'll be able to implement that due to some other projects that we have.

But this is definitely on my TODO list. Of course, you are welcome to help out through pull requests in case you manage to get this working before me :)