luxonis/depthai #483: [Feature-Request] Run multiple models in pipeline

franva, 2021-09-18:
### Start with the `why`:
Sometimes for a project, it requires more than 1 function e.g. segmentation and object tracking. 

### Move to the `what`:
For this reason I would like to see whether it's possible to run 2 models in a pipeline on the same OAK-D camera?

Is it possible?

If yes, do we have a demo?

Many thanks

Luxonis-Brandon, 2021-09-18:
Yes this is why we re-wrote our Gen1 API from scratch to make the Gen2 pipeline builder system that has been mainlined since the beginning of this year.  

https://github.com/luxonis/depthai/issues/136

There are many examples here:
https://github.com/luxonis/depthai-experiments

Some specific examples:
https://github.com/luxonis/depthai-experiments/tree/master/gaze-estimation
https://github.com/luxonis/depthai-experiments/tree/master/gen2-age-gender
https://github.com/luxonis/depthai-experiments/tree/master/gen2-face-recognition
https://github.com/luxonis/depthai-experiments/tree/master/gen2-human-machine-safety
https://github.com/luxonis/depthai-experiments/tree/master/gen2-license-plate-recognition
https://github.com/geaxgx/depthai_hand_tracker#body-pre-focusing

Thanks,
Brandon

franva, 2021-09-18:
Beautiful~!

Thanks @Luxonis-Brandon , I will have a look. For now, I will close this request as it seems finished.

closed 2021-09-18 by franva