luxonis/depthai #484: [BUG] ObjectTracker repeatedly removes and then reaquries tracks

JasonHMaxwell, 2021-09-19:
**Describe the bug**
For objects of any class other than the first detected class, the ObjectTracker pipeline node will emit a NEW tracklet followed by a REMOVED tracklet in the next frame. This behavior repeats over the entire sequence of frames, i.e. you'll see NEW, REMOVED, NEW, REMOVED, NEW, REMOVED... I have observed this behavior only with the YOLO pipeline node.

**To Reproduce**
To make this as easy to reproduce as possible, I updated the `object_tracker_video` example code to use YOLO + the yolov4 model zoo model instead of mobilenet. I then took the darknet sample image and turned it into a 4 second video (i.e. 100 frames of the same stationary image, see attached) so that the tracker's job is as easy as possible. The tracker has been configured to use the UNIQE_ID TrackerIdAssigmentPolicy, so you should see the id for the dog and horse detections go up with each frame. The ID of the person object will remain constant.

```python
#!/usr/bin/env python3

from pathlib import Path
import cv2
import depthai as dai
import numpy as np
import time
import argparse

labelMap = [
    "person",         "bicycle",    "car",           "motorbike",     "aeroplane",   "bus",           "train",
    "truck",          "boat",       "traffic light", "fire hydrant",  "stop sign",   "parking meter", "bench",
    "bird",           "cat",        "dog",           "horse",         "sheep",       "cow",           "elephant",
    "bear",           "zebra",      "giraffe",       "backpack",      "umbrella",    "handbag",       "tie",
    "suitcase",       "frisbee",    "skis",          "snowboard",     "sports ball", "kite",          "baseball bat",
    "baseball glove", "skateboard", "surfboard",     "tennis racket", "bottle",      "wine glass",    "cup",
    "fork",           "knife",      "spoon",         "bowl",          "banana",      "apple",         "sandwich",
    "orange",         "broccoli",   "carrot",        "hot dog",       "pizza",       "donut",         "cake",
    "chair",          "sofa",       "pottedplant",   "bed",           "diningtable", "toilet",        "tvmonitor",
    "laptop",         "mouse",      "remote",        "keyboard",      "cell phone",  "microwave",     "oven",
    "toaster",        "sink",       "refrigerator",  "book",          "clock",       "vase",          "scissors",
    "teddy bear",     "hair drier", "toothbrush"
]

nnPathDefault = str((Path(__file__).parent / Path('models/yolo-v4-tiny-tf_openvino_2021.4_6shave.blob')).resolve().absolute())
videoPathDefault = str((Path(__file__).parent / Path('models/construction_vest.mp4')).resolve().absolute())
parser = argparse.ArgumentParser()
parser.add_argument('-nnPath', help="Path to yolo detection network blob", default=nnPathDefault)
parser.add_argument('-v', '--videoPath', help="Path to video frame", default=videoPathDefault)

args = parser.parse_args()

# Create pipeline
pipeline = dai.Pipeline()

# Define sources and outputs
manip = pipeline.createImageManip()
objectTracker = pipeline.createObjectTracker()
detectionNetwork = pipeline.createYoloDetectionNetwork()

manipOut = pipeline.createXLinkOut()
xinFrame = pipeline.createXLinkIn()
trackerOut = pipeline.createXLinkOut()
xlinkOut = pipeline.createXLinkOut()
nnOut = pipeline.createXLinkOut()

manipOut.setStreamName("manip")
xinFrame.setStreamName("inFrame")
xlinkOut.setStreamName("trackerFrame")
trackerOut.setStreamName("tracklets")
nnOut.setStreamName("nn")

# Properties
xinFrame.setMaxDataSize(1920*1080*3)

manip.initialConfig.setResize(416, 416)
manip.initialConfig.setKeepAspectRatio(False)
# manip.initialConfig.setResize(384, 384)
# manip.initialConfig.setKeepAspectRatio(False) #squash the image to not lose FOV
# The NN model expects BGR input. By default ImageManip output type would be same as input (gray in this case)
manip.initialConfig.setFrameType(dai.ImgFrame.Type.BGR888p)
manip.inputImage.setBlocking(True)

[... 111 lines cut ...]
            try:
                label = labelMap[t.label]
            except:
                label = t.label

            cv2.putText(trackerFrame, str(label), (x1 + 10, y1 + 20), cv2.FONT_HERSHEY_TRIPLEX, 0.5, 255)
            cv2.putText(trackerFrame, f"ID: {[t.id]}", (x1 + 10, y1 + 35), cv2.FONT_HERSHEY_TRIPLEX, 0.5, 255)
            cv2.putText(trackerFrame, t.status.name, (x1 + 10, y1 + 50), cv2.FONT_HERSHEY_TRIPLEX, 0.5, 255)
            cv2.rectangle(trackerFrame, (x1, y1), (x2, y2), color, cv2.FONT_HERSHEY_SIMPLEX)

        cv2.putText(trackerFrame, "Fps: {:.2f}".format(fps), (2, trackerFrame.shape[0] - 4), cv2.FONT_HERSHEY_TRIPLEX, 0.4, color)

        cv2.imshow("tracker", trackerFrame)

        if cv2.waitKey(1) == ord('q'):
            break
```

**Expected behavior**
The dog and horse detections should each be assigned a track in the first frame and these tracks should last the entire video.

**Screenshots**
[image]

https://user-images.githubusercontent.com/91002185/133919361-52398f3c-ede0-4cef-870d-e218fad70178.mp4

Erol444, 2021-09-20:
Hello @JasonHMaxwell, sorry about the inconvenience. This shouldn't happen, but we, unfortunately, can't do anything about it. Object tracker logic is proprietary software of Intel and is compiled so we can't change/fix anything. There are also other problems with the current object tracker: 

one thing I notice is that tracklets get removed sometime after 1 "LOST" status and sometimes after 30. Not really sure how to explain that. I also believe that the threshold isn't (always) working as expected. Also there is some random issue that it will create multiple tracklets for the same object, link here (I'm not sure if it has been resolved)

So my suggestion would be to either use mobilenet (maybe better detection accuracy?) or run a custom tracking NN.

Thanks, Erik

JasonHMaxwell, 2021-09-22:
Thanks for your reply @Erol444. I've manged to work around this issue by pulling the nn detections back to the host, setting the labels all to the same value, e.g. 1, and then sending the modified detections to the on-device tracker. It's possible that throwing away the detection labels will reduce the accuracy of the tracker, but I'm not sure what algorithm is being used. Is the tracking algorithm documented somewhere?. If the algorithm doesn't use the image data at all, then I may as well do the tracking on the host.

Would you be able to raise a bug with intel? Having said that, it sounds like Intel's tracker is a bit of a disaster, so maybe it would be faster for you to implement your own.