luxonis/depthai #485: RuntimeError | Virtual Box unable to recognize the USB device

laudoluk, 2021-09-20:
After Installing all the prerequisites listed in the [First Steps](https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#first-steps-with-depthai) section of the Depth AI online database, I tried running the Demo Script: 
                          $ python3 depthai_demo.py

However I kept getting the Error:
                          Using depthai module from:  /home/laudo/depthai/myvenv/lib/python3.6/site-packages/depthai.cpython-36m-x86_64-linux-gnu.so
Depthai version installed:  2.10.0.0.dev+7a0749a61597c086c5fd6e579618ae33accec8df
Traceback (most recent call last):
  File "depthai_demo.py", line 113, in <module>
    device_info = conf.getDeviceInfo()
  File "/home/laudo/depthai/depthai_helpers/config_manager.py", line 280, in getDeviceInfo
    raise RuntimeError("No DepthAI device found!")
RuntimeError: No DepthAI device found!

In Virtual Box, in the USB section of the Machine settings, I can also see a an unchecked box for xHCI which is the USB3 controller, however I cannot check it for some unknown reason. also the only option it currently considers is OHCI (USB1.1). Please do let me know if you know a way around this.

Also, one thing worth metnioning is that I am using Ubuntu 18.04.5 and the instruction I saw [here](https://docs.luxonis.com/projects/api/en/latest/install/#virtual-box) appears to be using Ubuntu 20.04.5  .

Luxonis-Brandon, 2021-09-20:
Hi @laudoluk ,

Sorry about the trouble.  So Virtualbox requires some additional steps to get the various modalities (USB2, USB3, pre-boot and post-boot) to be passed through into Virtualbox. Out of respect for the host being able to access these devices, Virtualbox requires each to be specifically added.

See our tutorial [here](https://docs.luxonis.com/projects/api/en/latest/install/#virtual-box) on how to get this running.  

Thoughts?

Thanks,
Brandon

laudoluk, 2021-09-21:
I realized today that the problem was that display option I selected for Virtualbox hid the necessary toolbox. For this I deactivated the Scaled option in the View section. After running the demo code (along with a few tries) I was able to see a new device with the name of Intel Corporation Luxonis Device [0100] and I selected it. Also, to make this permanent I added this device in the USB Settings and applied a filter. Now whenever I run the demo code it automatically connects to the Loopback device. Works like a charm!

closed 2021-09-21 by laudoluk