luxonis/depthai #488: Why add min allowed SquareSize?

nalzok, 2021-09-23:
Just wondering what's the purpose of this commit: https://github.com/luxonis/depthai/commit/bcf503c11ef6ec9d23c5f930040908178fd4152c. The [calibration board](https://github.com/luxonis/depthai/blob/main/charuco_297x210_8x11_20_DICT_4X4.pdf) in the repository explicitly states "Checker Size: 20 mm", so why enforce a minimal allowed SqureSize of 22 mm? 🤔

Luxonis-Brandon, 2021-09-23:
So when the board is printed or displayed below 22mm - it is often too small in the view of the cameras to get a proper read.  And so then the calibration actually ends up invalid in a super-confusion way.  Digital aliasing (I suppose) results in a calibration with supposed low error, that is actually completely off in terms of actual depth values (as I think there is digital aliasing occurring).

Either way, using a board that is 22mm or below results in invalid depth.  We could use a different board though so it's less confusing.  That's a great point.

Thoughts @saching13 ?

saching13, 2021-09-23:
@nalzok  Sorry for the confusion. Like Brandon mentioned we got bad results with 20mm. Markers corner had hard time detecting with good accuracy when the board was not too close.  And we don't want the board to be too close to get better calibration.
And since the PDF is vectorized we can print it in the size we want. And the file was generated from calib.io and we should have replaced it but I missed it. Sorry. 
I will replace it right away.

VanDavv, 2021-09-25:
@nalzok PR is here - https://github.com/luxonis/depthai/pull/493