luxonis/depthai #495: [Feature-Request] {Add timestamps to Feature Tracker Output}

arthurBricq, 2021-09-29:
### Start with the `why`:

The **feature tracker node** does not provide the timestamps of the image where the features were extracted. In order to get them, one has to also fetch the images themselves, or to read the time at which the features are received. 

It may not be desired: for some algorithm, like motion estimation, only the features may be needed. Access to the real timestamps (and not the time at which the features are received) could increase performances ! 

### Move to the `what`:

Add to the [RawTrackedFeatures](https://github.com/luxonis/depthai-shared/blob/2bd2bb926ceb9f52969ab42f64689e79d067b311/include/depthai-shared/datatype/RawTrackedFeatures.hpp) class the timestamps, like it is the case for instance for the class [RawImgFrame](https://github.com/luxonis/depthai-shared/blob/2bd2bb926ceb9f52969ab42f64689e79d067b311/include/depthai-shared/datatype/RawImgFrame.hpp). 

I'm guessing it's probably really quick to do so, but I am quite unsure of where to do it ! 

Thanks in advance,
Cheers !

szabi-luxonis, 2021-09-29:
@arthurBricq thanks for the recommendation.
You can set `setMetadataOnly(true)` property for XlinkOut node to which `passthroughInputImage` is connected from `FeatureTracker` node, which will output only the metadata, e.g. timestamp, frametype etc. instead the whole image data + metadata.

arthurBricq, 2021-09-29:
Thanks for the fix ! 
I hadn't thought about it, for sure it will do the job :+1:

closed 2021-09-29 by arthurBricq