luxonis/depthai #501: [BUG] {Bullseye Running Example from the Start ** Oak-1 **}

silver2row, 2021-10-11:
**Check if issue already exists**
I looked online and in the error sections on this repo. for ideas about running this hardware and software on the Debian Bullseye desktop I have currently. 

...

Everything compiles correctly. I added the udev rules also and then removed the USB from the host and reinserted it. Notta.

**Describe the bug**
The first example in the docs. section does not run: ` python3 depthai_demo.py ` 

**To Reproduce**
Steps to reproduce the behavior:
1. ` python3 depthai_demo.py ` 
```
[0] 14442C1071C847D700 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Disabling depth...
No previews available, adding color...
Downloading /home/torcha/.cache/blobconverter/mobilenet-ssd_openvino_2021.4_6shave.blob...
```
After some info. about software packaging lines and errors, I get this output.

```
requests.exceptions.ConnectionError: HTTPConnectionPool(host='blobconverter.luxonis.com', port=80): Max retries exceeded with url: /compile?version=2021.4 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f7b67fd0c70>: Failed to establish a new connection: [Errno 110] Connection timed out'))
```

**Expected behavior**
I expected the demo to run as is...

**Screenshots**
I do not have any screenshots yet. 

**Attach system log**

```
./log_system_information.py 
Defaulting to user installation because normal site-packages is not writeable
Collecting pyusb
  Downloading pyusb-1.2.1-py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 2.0 MB/s 
Installing collected packages: pyusb
Successfully installed pyusb-1.2.1
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.10.0-9-amd64-x86_64-with-glibc2.31",
    "processor": "",
    "python_build": "default Feb 28 2021 17:03:44",
    "python_compiler": "GCC 10.2.1 20210110",
    "python_implementation": "CPython",
    "python_version": "3.9.2",
    "release": "5.10.0-9-amd64",
    "system": "Linux",
    "version": "#1 SMP Debian 5.10.70-1 (2021-09-30)",
    "win32_ver": "",
    "uname": "Linux pack 5.10.0-9-amd64 #1 SMP Debian 5.10.70-1 (2021-09-30) x86_64",
    "packages": [
        "argcomplete==1.12.1",
        "blobconverter==1.0.0",
        "boto3==1.18.58",
        "botocore==1.21.58",
        "Brlapi==0.8.2",
        "certifi==2020.6.20",
        "chardet==3.0.4",
        "chrome-gnome-shell==0.0.0",
        "cupshelpers==1.0",
        "cycler==0.10.0",
        "dbus-python==1.2.16",
        "depthai==2.10.0.0.dev0+7a0749a61597c086c5fd6e579618ae33accec8df",
        "-e git+https://github.com/luxonis/depthai.git@2d13ccf734eba8b4ffd5e5845fef0f966f52db3f#egg=depthai_sdk&subdirectory=depthai_sdk",
        "distro==1.5.0",
        "distro-info==1.0",
        "feedparser==5.2.1",
        "ffmpy3==0.2.4",
        "httplib2==0.18.1",
        "idna==2.10",
        "jmespath==0.10.0",
        "kiwisolver==1.3.1",
        "louis==3.16.0",
        "Mako==1.1.3",
        "MarkupSafe==1.1.1",
        "matplotlib==3.3.4",
        "musicbrainzngs==0.7.1",
        "mutagen==1.45.1",
        "numpy==1.19.5",
        "olefile==0.46",
        "opencv-contrib-python==4.5.1.48",
        "opencv-python==4.5.1.48",
        "Pillow==8.1.2",
        "pip==21.2.4",
        "pycairo==1.16.2",
        "pycups==2.0.1",
        "pycurl==7.43.0.6",
        "PyGObject==3.38.0",
        "pyinotify==0.9.6",
        "pyparsing==2.4.7",
        "PySimpleSOAP==1.16.2",
        "pysmbc==1.0.23",
        "python-apt==2.2.1",
        "python-dateutil==2.8.1",
        "python-debian==0.1.39",
        "python-debianbts==3.1.0",
        "pytube==11.0.1",
        "pyusb==1.2.1",
        "pyxdg==0.27",
        "PyYAML==5.4.1",
        "quodlibet==4.3.0",
        "reportbug==7.10.3+deb11u1",
        "requests==2.24.0",
        "s3transfer==0.5.0",
        "setuptools==52.0.0",
        "six==1.16.0",
        "unattended-upgrades==0.1",
        "urllib3==1.25.11",
        "wheel==0.34.2",
        "xdg==5"
    ],
    "usb": [
        {
            "port": 1,
            "vendor_id": "0x8087",
            "product_id": "0x8000",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x8087",
            "product_id": "0x8008",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0003",
            "speed": "Super"
        },
        {
            "port": 7,
            "vendor_id": "0x0bda",
            "product_id": "0x0184",
            "speed": "High"
        },
        {
            "port": 4,
            "vendor_id": "0x046d",
            "product_id": "0xc52f",
            "speed": "Full"
        },
        {
            "port": 13,
            "vendor_id": "0x0cf3",
            "product_id": "0xe004",
            "speed": "Full"
        },
        {
            "port": 12,
            "vendor_id": "0x03e7",
            "product_id": "0x2485",
            "speed": "High"
        },
        {
            "port": 1,
            "vendor_id": "0x046d",
            "product_id": "0xc318",
            "speed": "Full"
        },
        {
            "port": 0,
            "vendor_id": "0x1d6b",
            "product_id": "0x0002",
            "speed": "High"
        }
    ]
}
System info gathered successfully - saved as "log_system_information.json"
```
**Additional context**
If you need any additional info, please reply. I cannot wait to get started using this camera.

silver2row, 2021-10-11:
Hello Again,

I would like to add something here. ` Available platform plugins are: xcb. ` happens when I switch branches to gen1_main.

Seth

P.S. If you get bored, let me know if you know my issue. In the meantime, I will try to look around online and in the chatroom at Discord.

alex-luxonis, 2021-10-11:
@silver2row Sorry for the trouble here...
We have a problem with https://blobconverter.luxonis.com that is being investigated.
For now, the `blobconverter` pip package was updated on `main`, so downloading at least some models works, but the default one `mobilenet-ssd` still fails: PR https://github.com/luxonis/depthai/pull/502 (merged).

Could you pull the latest from the `main` branch, and run this:
```
python3 install_requirements.py
python3 depthai_demo.py -cnn face-detection-retail-0004
```

silver2row, 2021-10-11:
Hello,

I am back on Win 10 right now. It is late over here. I will get to it tomorrow. Sorry for the inconvenience. I will test ` main ` with those commands.

Seth

P.S. @alex-luxonis : Thank you for clarifying so quickly. I will get on it soon.