luxonis/depthai #518: [BUG] error message: X_LINK_DEVICE_NOT_FOUND

tottanka, 2021-11-01:
While trying to first run our new OAK D POE i seem to be unable to run the depthai demo. I was able to run the frame demo earlier though, but once i try the depthat.py i get the following:
`sing depthai module from:  c:\Users\Liron\anaconda3\envs\OakD\venv\lib\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.10.0.0.dev+7a0749a61597c086c5fd6e579618ae33accec8df
Available devices:
[0] 169.254.1.222 [X_LINK_BOOTLOADER]
Traceback (most recent call last):
  File "c:\Users\Liron\anaconda3\envs\OakD\depthai_demo.py", line 140, in <module>
    with dai.Device(pm.pipeline.getOpenVINOVersion(), deviceInfo, usb2Mode=conf.args.usbSpeed == "usb2") as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND`

I am running VS Code on a windows 10 computer with the camera plugged in to the ethernet port whit a POE injector.
I have tried turning off Wifi and all firewalls and still get the same result.
I have even tried installing everything on a different PC just to get the same error.

drobb257, 2021-11-02:
I'm seeing something very similar on Ubuntu 20 over USB when trying to run the depthai_demo.py

`
Depthai version installed:  2.10.0.0.dev+7a0749a61597c086c5fd6e579618ae33accec8df
Available devices:
[0] 14442C10C19E3DD700 [X_LINK_UNBOOTED]
Traceback (most recent call last):
  File "depthai_demo.py", line 140, in <module>
    with dai.Device(pm.pipeline.getOpenVINOVersion(), deviceInfo, usb2Mode=conf.args.usbSpeed == "usb2") as device:
RuntimeError: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

`

I've tried installing different versions of the library, as well as testing both an OAK-D and OAK-1.   All have resulted in the same error message.  Any ideas on what to try?

Luxonis-Brandon, 2021-11-02:
For the USB case, it is very likely this:
https://docs.luxonis.com/en/latest/pages/troubleshooting/#failed-to-find-device-ma2480-error-message-x-link-device-not-found
To fix this, set the udev rules using the commands below, unplugging DepthAI and then plugging it back into USB afterwards.

echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
And in some cases, these were already set, but DepthAI was plugged in the entire time, so Linux could not reset the rules.

On the PoE issue - it is likely unrelated.  Asking the team.

alex-luxonis, 2021-11-02:
@tottanka I see the device is found on the link-local IP address `169.254.1.222`, which is expected in most cases for direct connections, no DHCP server.
Windows assigns to itself an address in the same range, but this happens after some time after the link-up event, when a timeout looking for DHCP expires.
Likely the issue happens because we have a link-down -> link-up transition when the DepthAI application is started, and the firmware is booted up. Then Windows starts again to look for DHCP, and some timeouts are triggered, both in the DepthAI library, and the watchdog in firmware (as the host didn't connect in time).

Could you try to assign a static IP in the same range to the network interface, like: `169.254.1.10`, subnet mask `255.255.0.0`, as in this picture:
[image]

(taken from https://avtech.com/articles/11155/assign-static-ip-address-windows-computer/)

drobb257, 2021-11-02:
@Luxonis-Brandon I've tried the udev rules, doesn't seem to help.  I was getting a different error before I fixed that, so I don't think that's the problem.

It also appears like the device is being detected:
`Available devices:
[0] 14442C10C19E3DD700 [X_LINK_UNBOOTED]
`

I also hear an audible click (relay or something?) from the device when I run the script, so it appears that some form of communication is happening?

Luxonis-Brandon, 2021-11-02:
So the click is the firmware being able to boot - to autofocus initializing.  And then the incapability to communicate with the device I think is the udev rules or some other issue.

Did you unplug and replug the device after doing the udev rules?  Also you could force USB2 in case it is a USB3 issue.

alex-luxonis, 2021-11-02:
Yeah, the udev rules aren't needed on Windows, but just on Linux

So, as Brandon said above, try running with:
`python depthai_demo.py -usbs usb2`

drobb257, 2021-11-02:
@alex-luxonis Forcing USB2 fixed it, thanks for your help.

alex-luxonis, 2021-11-02:
Great! But it's still not optimal, the throughput could be limited, depending on usage (what streams are enabled). 
Are you using the original USB3 cable, shipped with the device?

Also try different USB ports on the PC, if available. If still not working, maybe try rebooting the PC. 

When an USB3 connection is established, depthai_demo.py will print:
`USB Connection speed: UsbSpeed.SUPER`

drobb257, 2021-11-02:
@alex-luxonis Yeah, I'm using the original cable that came with the device.  I'm sure that both computers I tested with are USB3 compatible, and I tested all ports on both computers.

ahmadchalhoub, 2023-02-09:
> For the USB case, it is very likely this: https://docs.luxonis.com/en/latest/pages/troubleshooting/#failed-to-find-device-ma2480-error-message-x-link-device-not-found To fix this, set the udev rules [...]

Any updates on what might be causing this on OAK-D cameras? I am currently getting this error as well.

SohailSayed, 2024-01-23:
@ahmadchalhoub Did you resolve this? Currently facing this issue with USB connection