luxonis/depthai #521: What is the best way to integrate Custom Detection + Tracking + Depth/Spatial Location Calculation

AlexanderJGomez, 2021-11-04:
Hello I am currently trying to use YOLOv5 with multiple object tracking (specifically SORT), retrieve the spatial coordinates and then save the associated depth frame.

I have used some of the code from https://github.com/luxonis/depthai-experiments/tree/master/gen2-yolov5 for decoding the detections from YOLOv5 and then passed that to my MOT to get bounding boxes associated with the tracked objects. All of this so far has worked well. I have used the depthai_demo.py code and inserted my own snippets to make it work, and I can verify it via the previews. 

Then, I've tried using SpatialLocationCalculator, similar to how it was done here https://docs.luxonis.com/projects/api/en/latest/samples/SpatialDetection/spatial_location_calculator/#spatial-location-calculator . However it seems like there is a frame or two of lag between enqueue'ing the ROIs and when I can actually read those same ROIs from the "spatialData" queue.

I have tried setting the `SpatialLocationCalculator`

`slc.setWaitForConfigInput(True)` and using 

`spatialDataQueue.get()`

And that fixes the ordering of messages received but seems to cause issues with reading new depth frames, which I can see in the depth preview window and also in the end print statement, where TOTAL FPS will usually show `color` FPS much higher than `depth` or `depthRaw`.  The `color` stream looks to be unchanged, though. It seems to me that the `setWaitForConfigInput(True)` also pauses the depth stream while it waits for a config input... is this expected behavior?

Also is there a better way for me to do this?

AlexanderJGomez, 2021-11-05:
Resolved the issue with output from SpatialLocationCalculator lagging the input, seems the input has a default queue size of 4.

Doing the following worked for me:

```
nodes.slc.inputConfig.setQueueSize(1)            
nodes.slc.setWaitForConfigInput(False)

```

AlexanderJGomez, 2021-11-05:
How can I make it so I send a SpatialLocationCalculatorConfig into the Calculator and when I read from that queue directly after I get the same SpatialLocationCalculatorConfig?

I've tried setting the queue sizes to 1 and non-blocking 
```python
spatialCalcQueue = device.getOutputQueue(name="spatialData", maxSize=1, blocking=False) 
spatialCalcConfigInQueue = device.getInputQueue("spatialCalcConfig", maxSize=1, blocking=False)
```
Here is some example output from my code, with the frame, whether I am sending to the inputConfig queue or reading from the spatialData out queue, and the width and height of the ROI (couldn't find some sort of ID attribute)

The first read is of the initial config

[image]

szabi-luxonis, 2021-11-05:
> How can I make it so I send a SpatialLocationCalculatorConfig into the Calculator and when I read from that queue directly after I get the same SpatialLocationCalculatorConfig?
>
> I've tried setting [...]

If you want to preserve ordering then you can do the following:
 ```
 setWaitForConfigInput(True)
 while True:
     send config data
     read output of spatial location calculator
 ```
This is the "synchronous" way, and will result in heavy performance drops since everything runs one after the other in a blocking manner.

szabi-luxonis, 2021-11-05:
> Hello I am currently trying to use YOLOv5 with multiple object tracking (specifically SORT), retrieve the spatial coordinates and then save the associated depth frame.

You can keep the same flow, without `setWaitForConfigInput` , `SpatialLocationCalculator` has a `passthroughDepth` [output](https://docs.luxonis.com/projects/api/en/latest/components/nodes/spatial_location_calculator/), which is the associated depth frame with the input image/config. You can use that to associate with spatial coordinates.

>`setWaitForConfigInput(True)` also pauses the depth stream while it waits for a config input... is this expected behavior?
Yes, it means that it waits for the input config message until it arrives, thus the rest of the inputs are blocked.

AlexanderJGomez, 2021-11-05:
Thanks for the response!

Is there any sort of ID that can be used to associate the read config message and the configdata back to the one that was sent? I would like to tie the spatial coords I read back to the detections

szabi-luxonis, 2021-11-05:
> Thanks for the response!
> 
> Is there any sort of ID that can be used to associate the read config message and the configdata back to the one that was sent? I would like to tie the spatial coords I read back to the detections

There isn't unique ID specific to config data, but there are a couple of ways around it:
1. You can store the sent config data in a `dictionary` , and check if the received data is in the dictionary.
2. It's hacky, but you can use `upperThreshold` (or `lowerThreshold`) to store a unique ID. (or the ROI etc.)
E.g. if you set it to `config.depthThresholds.upperThreshold = 10000` then you can increment for each new config, and reset back to the initial value after let's say `50` new IDs.
3. You can use `Script` node to eliminate all host side interaction, and eventually add a custom message that stores ID.
This is the way to go, but more complicated, since you would need to move all logic from host to device.

I'd go with option 2, hacky, but the most trivial to implement.

AlexanderJGomez, 2021-11-05:
If I wanted to option 2 that would mean implementing all of the logic used in the SpatialLocationCalculator node, right?

Might it work to just do all of the calculation of the spatial location on the host? Taking the detections and the depth frame and calculating the depth directly? Like is done here: https://github.com/luxonis/depthai-experiments/blob/3c2c0d9fa3d33ac18a05701604c9d889048a3a64/gen2-human-machine-safety/main.py#L47

And lastly, how can I correctly align the bounding boxes I pass in from the detection done on the color camera with the correct location on the depth frame?

szabi-luxonis, 2021-11-05:
> Might it work to just do all of the calculation of the spatial location on the host? Taking the detections and the depth frame and calculating the depth directly? Like is done here: [...]

Yes, you can use that.

> And lastly, how can I correctly align the bounding boxes I pass in from the detection done on the color camera with the correct location on the depth frame?

You can align depth frame to RGB camera, like here:
https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/rgb_depth_aligned.py#L51

If depth is aligned to RGB then you don't need to reproject the bounding box.
The only limitation is that camera output needs to be the same resolution as depth frame, e.g. in case of 1080P RGB and 720P depth you need to set `camRgb.setIspScale(2, 3)` too.

AlexanderJGomez, 2021-11-05:
Sorry I think I'm a bit confused. It seems that for the `depthai_demo.py` we never use

`stereo.setDepthAlign`

does this mean that for that example the bounding boxes we see are not actually properly aligned?

From what I understand, we have the following image details:

# Color Camera

We get a, by default, 1920x1080p image from the color camera which is resized by, I assume, just changing the aspect ratio and without any clipping of the image(in my example to 416x416) as input to the NN from camRGB.preview [here](https://github.com/luxonis/depthai/blob/main/depthai_sdk/src/depthai_sdk/managers/nnet_manager.py#L173). Seems like that is the case for the image manipulations because of the explanation [here](https://docs.luxonis.com/projects/api/en/latest/tutorials/maximize_fov/) when setPreviewKeepAspectRatio(False).

Then, to display the predicted bounding boxes we scale to the size of the output from the "video" stream, as it appears [here](https://github.com/luxonis/depthai/blob/main/depthai_sdk/src/depthai_sdk/managers/pipeline_manager.py#L135) and render them onto that frame. For some reason, my "video" stream is `(399, 710, 3)`, which does appear to be the same aspect ratio as 1080x1920

# Depth Camera

defaults to 640x400, so has a different asp ratio, and with a different FOV as well.

I'm confused how setting depthAlign and IspScale accomplishes aligning these two different camera settings

szabi-luxonis, 2021-11-06:
@AlexanderJGomez 

>does this mean that for that example the bounding boxes we see are not actually properly aligned?

Yes.

The bounding boxes which are obtained from `preview` output of RGB camera are not aligned properly with `depth` frame because that is aligned to `rectified right` perspective by default (right stereo camera). The only alignment that is done is scaling the bounding box, taking into consideration FOV difference between RGB and right stereo camera.
To obtain proper alignment you need to use `setDepthAlign` and `setIspScale` option as mentioned before, which will align the depth frame to RGB camera perspective, so bounding boxes will have proper aligment.

AlexanderJGomez, 2021-11-10:
Thanks a ton for the responses!

I am now at the point of implementing the spatials calculation using the code from [here](https://github.com/luxonis/depthai-experiments/blob/3c2c0d9fa3d33ac18a05701604c9d889048a3a64/gen2-human-machine-safety/main.py#L47). However, after doing some testing of my results to that of the SpatialLocationCalculator, it seems like my results are a bit off. 

I have made edits to the code [here](https://github.com/luxonis/depthai-experiments/blob/3c2c0d9fa3d33ac18a05701604c9d889048a3a64/gen2-human-machine-safety/main.py#L76) to fit my rectangular image, and those changes look like the following:

I've changed to:

```python

    midX = int(depth.shape[1] / 2)
    midY = int(depth.shape[0] / 2) # middle of the depth img
    bb_x_pos = centroidX - midX
    bb_y_pos = centroidY - midY

    angle_x = calc_angle(bb_x_pos, depth.shape[1])
    angle_y = calc_angle(bb_y_pos, depth.shape[0])
```

and added a param to `calc_angle` so the code is changed from [this](https://github.com/luxonis/depthai-experiments/blob/3c2c0d9fa3d33ac18a05701604c9d889048a3a64/gen2-human-machine-safety/main.py#L158) to:

```python
def calc_angle(offset, num_pixels):
    return math.atan(math.tan(monoHFOV / 2.0) * offset / (num_pixels / 2.0))
```

I have tried to compare my calculations to the results from `SpatialLocationCalculator` using the `passthroughDepth` frame from the node, however it seems like my calculations are slightly off and I'm not sure why. I also did make sure to pass only the scaled size roi to SLC to match what is done in calc_spatials

Are there any changes that need to be made to 
```python
monoHFOV = np.deg2rad(73.5)
``` 
I noticed some cropping of the image being done in `crop_to_rect` and the line `cam.initialControl.setManualFocus(130)` that is being run in the referenced code and was not sure if that affected anything.
?

szabi-luxonis, 2021-11-15:
@AlexanderJGomez 
`calc_spatials` uses the mean value in ROI while on-device SLC uses the average.
Here is the equivalent of SLC algorithm on host:
https://github.com/luxonis/depthai-experiments/blob/master/gen2-spi/spatial-location-calculator/main.py#L150

closed 2021-11-19 by AlexanderJGomez