luxonis/depthai #541: [BUG] Oak-D-PoE on Mac and LAN, is visible, but produces [X_LINK_BOOTLOADER] error

mpurnyn, 2021-11-24:
**Check if issue already exists**
I checked and nothing looks similar enough to my circumstances. 

**Describe the bug**
I got a new Oak-D-PoE and I wanted to try to connect to it with the demo code. In the past I have managed to get the OAK-D and OAK working on Mac, Windows, and Linux over USB however get this error when trying to do the same over Ethernet. I first tried to connect directly to the PC and I managed to ping it, but wasnt able to find it automatically, but then I tried to do it over LAN and It discovers it, but produces this error.

`(luxonis) ppurnyn@JP-KIT-M-232 depthai % python3 depthai_demo.py 
Using depthai module from:  /Users/ppurnyn/Library/Python/3.9/lib/python/site-packages/depthai.cpython-39-darwin.so
Depthai version installed:  2.13.0.0
Setting up demo...
Available devices:
[0] 192.168.1.112 [X_LINK_BOOTLOADER]
Stack trace (most recent call last) in thread 12920614912:
#11   Object "libsystem_pthread.dylib", at 0x7ff811701513, in _pthread_start + 124
#10   Object "QtCore", at 0x163e556a7, in QTimerInfoList::activateTimers() + 5463
#9    Object "QtQuick", at 0x16a114b27, in QNativeInterface::QSGMetalTexture::fromNative(id<MTLTexture>, QQuickWindow*, QSize const&, QFlags<id<MTLTexture>::CreateTextureOption>) + 28471
#8    Object "QtCore", at 0x163e3c989, in QMacAutoReleasePool::~QMacAutoReleasePool() + 41
#7    Object "Foundation                          0x00007ff812528057", at 0x7ff812528057, in drain] + 132
#6    Object "CoreFoundation", at 0x7ff8117895df, in _CFAutoreleasePoolPop + 21
#5    Object "libobjc.A.dylib", at 0x7ff81159939f, in objc_autoreleasePoolPop + 160
#4    Object "libobjc.A.dylib", at 0x7ff81159c41e, in AutoreleasePoolPage::releaseUntil(objc_object**) + 166
#3    Object "libobjc.A.dylib", at 0x7ff811597e80, in objc_release + 32
#2    Object "libsystem_platform.dylib", at 0x7ff811716e2c, in _sigtramp + 28
#1    Object "depthai.cpython-39-darwin.so", at 0x11b04b5fd, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 13
#0    Object "depthai.cpython-39-darwin.so", at 0x11b04b66c, in backward::SignalHandling::sig_handler(int, __siginfo*, void*) + 124
zsh: segmentation fault  python3 depthai_demo.py
(luxonis) ppurnyn@JP-KIT-M-232 depthai % python depthai_demo.py 
  File "depthai_demo.py", line 42
    Trackbars.instances[name] = {**Trackbars.instances.get(name, {}), window: defaultVal}
                                  ^
SyntaxError: invalid syntax
`

**To Reproduce**
Steps to reproduce the behavior:
1. Run script python3 depthai_demo.py on Mac M1
2. See error

**Expected behavior**
demo connects and produces no error

alex-luxonis, 2021-11-24:
@mpurnyn I just tried running the demo with an OAK-D-PoE on a Mac mini M1 with Big Sur 11.1, Python3.9, with the first command you tried `python3 depthai_demo.py` and it works fine.
(For the second command, `python` seems to point to `python2` issuing that invalid syntax error.)

I'm installing in parallel Monterey 12.0.1 to test there as well.

The error seems related to our recent Qt6 based GUI addition to the demo. Could you try with this command instead:
`QT_QUICK_BACKEND=software python3 depthai_demo.py`

If the issue persists, just for test, try downgrading to an older version, before the GUI addition:
```
git fetch
git checkout old_interface_before_gui
```

Also, was the dependencies installation on the M1 done using the command mentioned here?
https://docs.luxonis.com/projects/api/en/latest/install/?#macos

mpurnyn, 2021-11-24:
Thank you for your quick response.

1.  I did miss the dependencies for m1 from your link, I ran the curl command and installed. It did end up erroring out
`HEAD is now at 3ab140e39 Merge pull request #12470 from Homebrew/dependabot/bundler/Library/Homebrew/sorbet-0.5.9368
==> Tapping homebrew/core
remote: Enumerating objects: 1097944, done.
remote: Counting objects: 100% (47/47), done.
remote: Compressing objects: 100% (29/29), done.
remote: Total 1097944 (delta 27), reused 34 (delta 18), pack-reused 1097897
Receiving objects: 100% (1097944/1097944), 419.61 MiB | 9.04 MiB/s, done.
Resolving deltas: 100% (756237/756237), done.
From https://github.com/Homebrew/homebrew-core
 * [new branch]              master     -> origin/master
HEAD is now at 5c075f4a626 php: remove `glib` dependency (#89902)
Error: Failed to link all completions, docs and manpages:
  Permission denied @ rb_file_s_symlink - (../../../Homebrew/completions/zsh/_brew, /usr/local/share/zsh/site-functions/_brew)
Failed during: /usr/local/bin/brew update --force --quiet`

2. I tried to run with the QT parameter provided and it worked after loading the camera.

3. I then tried to downgrade to the older version and that worked as well.

4. My Mac is currently running 12.0.1

I guess I should still be able to connect and use the camera, even if QT and the demo do not work, right?

alex-luxonis, 2021-11-24:
Yes, QT is actually used just with this demo. All other examples should work:
https://github.com/luxonis/depthai-python/tree/main/examples

Good to know that applying the `QT_QUICK_BACKEND=software` environment variables resolves the problem. 
Actually we saw a similar issue on Windows and applied it directly inside the demo: https://github.com/luxonis/depthai/pull/532/files
The easy workaround for us would be to do the same for macOS, but since the above might slightly affect the performance (e.g. increase CPU load on host), it's better to look for a proper solution. I'm checking on a freshly installed Monterey as well now.

About downgrading, thanks for the test as well. Note: don't forget to check out back the `main` branch for the latest.

Will check as well installing the dependencies. It's possible it failed for you because Homebrew was already installed.

alex-luxonis, 2021-11-24:
Ran `bash -c "$(curl -fL http://docs.luxonis.com/_static/install_dependencies.sh)"`
on a freshly installed Monterey 12.0.1 on a Mac mini M1, then cloned this repo, ran `install_requirements.py` and `depthai_demo.py` and it worked all good without setting the environment variable (`main` branch, not on the PR mentioned above).

So it could be that it depends on whether other software packages are (or are not) installed in the system. 
Anyway, I checked the CPU load with `top`, and didn't see a noticeable difference when having the variable set or not. So we'll add a workaround for now: https://github.com/luxonis/depthai/pull/544

mpurnyn, 2021-11-25:
Thanks for putting in the fix. I managed to get another error when going back to main though. It does not appear when using the before-gui tag though. I went back to the older tag for now.

`env-depthai) ppurnyn@JP-KIT-M-232 depthai % QT_QUICK_BACKEND=software python3 depthai_demo.py -cnn person-detection-retail-0013
Using depthai module from:  /Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/env-depthai/lib/python3.9/site-packages/depthai.cpython-39-darwin.so
Depthai version installed:  2.13.0.0
Setting up demo...
Available devices:
[0] 192.168.1.112 [X_LINK_BOOTLOADER]
  File "/Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/depthai/depthai_demo.py", line 549, in run
    self.instance.run_all(self.conf)
  File "/Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/depthai/depthai_demo.py", line 60, in run_all
    self.setup(conf)
  File "/Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/depthai/depthai_demo.py", line 135, in setup
    self.metrics.reportDevice(self._device)
  File "/Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/depthai/depthai_helpers/metrics.py", line 53, in reportDevice
    self.demo_table.update({
  File "/Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/env-depthai/lib/python3.9/site-packages/pyrebase/pyrebase.py", line 309, in update
    raise_detailed_error(request_object)
  File "/Users/ppurnyn/workspace/kws-vision/kpmgjp-kit-lab-kws-vision/env-depthai/lib/python3.9/site-packages/pyrebase/pyrebase.py", line 448, in raise_detailed_error
    raise HTTPError(e, request_object.text)
[Errno 400 Client Error: Bad Request for url: https://depthai-data-default-rtdb.firebaseio.com/demo.json] {
  "error" : "Invalid data; couldn't parse JSON object. Are you sending a JSON object with valid key names?"
}`

alex-luxonis, 2021-11-25:
(reopening automatically closed issue)

@mpurnyn Thanks for testing, I merged the above PR to `main`, so manually prepending `QT_QUICK_BACKEND=software` should no longer be needed.

And sorry about the new issue, seems it's introduced by https://github.com/luxonis/depthai/pull/531 (merged yesterday). I get the same failure with an OAK-D-PoE, but works with OAK-D (over USB).

We'll work on fixing it.

alex-luxonis, 2021-11-25:
I also edited your comment above to enclose the log snippet within triple backticks ` ``` multi-line snippet ``` `, as apparently it's not displayed properly by single backticks, line terminators were ignored.

alex-luxonis, 2021-11-25:
@mpurnyn Should be fixed by https://github.com/luxonis/depthai/pull/548 (merged to `main`).