luxonis/depthai #546: [BUG] Consistent performance difference between PoE and USB connected devices

wrlear, 2021-11-24:
**Describe the bug**
When running the same TinyYOLOV4 detection model, The OAK-D-PoE shows the LeonOS CPU load as 90-93% while the OAK-D and OAK-D-LITE show a LenoOS CPU load of only 48-50%. 

The measured NN FPS for the PoE camera is consistently 15-20% lower than the USB connected camera: 719-20fps vs22-23fps.

**To Reproduce**
The script I'm running is derived from the "spatial_tiny_yolo.py" example script. Per discussions with @Luxonis-Brandon , I disabled all image retrieval and display in order to rule out differences between ethernet and USB data rates - the script just retrieves the detections, computes a running average of fps, and prints it to the command line.  
The CPU load and performance differences do not appear to be significantly affected by the number of objects detected.

**Expected behavior**
I would expect the performance of the two systems to be within a few percent of each other. With the existing PoE CPU load, I am concerned about the ability to add on-device scripting to my application

brmarkus, 2021-11-25:
Can a HW-expert shed some light into what the differences are between the data being transferred via USB (USB3 superspeed?) and Ethernet?
Does the used Ethernet-controller has the same USB-interface(s) to transfer the data? Is I2C/I3C involved as additional "protocol"?
Is DMA being used, paging? Interrupt-driven?
Is compression or other proprietary protocols (encryption&decryption) applied and causing higher system-load?

Luxonis-Brandon, 2021-11-27:
Great question.  The main difference is the use of PCIE.  The PoE models act as a PCIE host, where the Myriad X is the host to a PCIE Ethernet MAC/PHY.  And in this configuration of the Myriad X (when the MX is in PCIE host mode) there is a limitation on the use of internal cache that interconnects the network-on-chip fabric.  This limitation results in not being able to use the cache as efficiently, which then results in more load on the LeonOS CPU.  

And this is why the PoE models see higher CPU use and lower performance - because this high-speed cache (as of now) cannot be properly utilized when the Myriad X is in PCIE host mode (which it has to be for PoE models).

Thoughts?

Thanks,
Brandon