luxonis/depthai #547: [BUG] X_LINK_ERROR and program abort during extended tests

wrlear, 2021-11-24:
**Describe the bug**
While running extended tests of the same TinyYOLOV4 script, the OAK-D-POE aborted with an X_LINK_ERROR after 7 hours, and the OAK-D-IOT (connected via USB to host) aborted after 49 hours with a similar error.

The PoE error was: 'Couldn't read data from stream: 'boundingBoxDepthMapping' (X_LINK_ERROR)'
The IOT error was:  'Couldn't read data from stream: 'detections' (X_LINK_ERROR)'
Note: an OAK-D-LITE running the same test ran for over 72 hours with no error.

All systems were connected to the same host PC with the python scripts running in separate shells, and no other interactive activity on the PC. Failed systems resumed normal operation when the scripts were restarted, but the OAK-D-POE aborted again after 8 hours.

During these tests, the LeonOS CPU load was significantly higher on the PoE device (90-92%) than on the USB device (49-50%). This has been reported in issue #546.

**To Reproduce**
Test script is described in issue #546 

**Expected behavior**
Assuming stable temperature, power, and host connections, I would expect these tests to run indefinitely.

**Additional context**
Note: in the initial test, the OAK-D-IOT reported internal temperatures in the range of 88-89C. To eliminate the possibility of a thermal issue, I restarted the test with an external fan directed at the heatsink, lowering the reported temperatures to 64-65C. It failed with the X_LINK_ERROR after 37 hours.

wrlear, 2021-11-24:
I overlooked the fact that the test script is also loading a "hello world" on-device script prior to entering the forever loop:

```
import time
count = 0
time.sleep(1)
while True:
    count = count + 1
    node.info(f"Hello from the on-device script {count}" )
    time.sleep(60)
```

Luxonis-Brandon, 2021-11-24:
Thanks for documenting this on Github @wrlear (from our discussion on Discord).  I think this has to do with some instability we are seeing with the scripting node on-device.  We are debugging.

Luxonis-Brandon, 2021-11-24:
Oh, and one quick question/test @wrlear would be to repeat the test not running anything on the script node at all.  As from previous similar situation we think the instability was coming from the scripting node itself.

wrlear, 2021-11-24:
@Luxonis-Brandon I just started that test on 2 OAK-D-IOT cameras, 1 OAK-D-IOT, and 1 OAK-D-LITE.  
No on-device scripting, No fetch or display of RGB image or Depth map data, only reading detections and printing statistics to cmdline.  LeonOS cpu loads are slightly reduced from previous tests:
PoE: LeonOS=85-89%,  NN FPS = 19-20
USB: LeonOS=45-47%, NN FPS = 22-23

The system information logs are being displayed every 60 seconds to give me a good indication of run time

wrlear, 2021-11-24:
I'm afraid one of the OAD-D-IOT cameras hit the error after only 21 minutes

```
[192.168.50.113] [1276.742] [system] [info] Memory Usage - DDR: 68.02 / 358.65 MiB, CMX: 2.24 / 2.50 MiB, LeonOS Heap: 63.96 / 78.59 MiB, LeonRT Heap: 5.11 / 23.84 MiB
[192.168.50.113] [1276.742] [system] [info] Temperatures - Average: 49.36 ┬░C, CSS: 50.03 ┬░C, MSS 49.14 ┬░C, UPA: 50.03 ┬░C, DSS: 48.25 ┬░C
[192.168.50.113] [1276.742] [system] [info] Cpu Usage - LeonOS 87.06%, LeonRT: 27.69%
... ~35 seconds later...
Traceback (most recent call last):
  File "C:\...\src\proto\proto.py", line 199, in <module>
    inDet = detectionNNQueue.get()
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'detections' (X_LINK_ERROR)'
```

I restarted that test

themarpe, 2021-11-24:
Hi @wrlear - just for sake of having identical setups, can you supply the copy of the script that you are using for testing (latest one without on-device scripting)
I"ll check whether the issue is the same as we observed with scripting (memory corruption in FW) and will dig into

wrlear, 2021-11-24:
I can send the script directly to you (how?). but the blob file cannot really be shared. However, it should exhibit the same behavior with the Tiny YOLOV4 model used in the "spatial_tiny_yolo.py" example.

themarpe, 2021-11-24:
@wrlear feel free to post it here as a codeblock (insert code, or triple backtick - code - triple backtick). And no problem regarding the model - will test with TinyYOLOv4

imnileshd, 2021-11-30:
Hey, 

how to handle below error?
RuntimeError: Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'video' (X_LINK_ERROR)'

Device - OAK-D-POE
Host - Windows 

To reproduce, I have attached code..
[main_issue.zip](https://github.com/luxonis/depthai/files/7628531/main_issue.zip)

Thanks,
Nilesh

wrlear, 2021-11-30:
FYI, I started Nilesh's script on one of my OAK-D-POEs and will let it run overnight - or until failure :-)

Erol444, 2021-12-01:
From wrl [on Discord](https://discord.com/channels/790680891252932659/795742702154153994/915704597702656070), this pipeline crashed after 4.5h. Schema attached.
[Schema.zip](https://github.com/luxonis/depthai/files/7637342/Schema.zip)

wrlear, 2021-12-03:
This may be just a coincidence, but I checked out the oak-d-pro-develop branch so I could include the OAK-D-PRO camera in my tests. For consistency, I am running all cameras on that version (2.13.2.0.dev0+164ad70f9ba08306d0a035824771bac28a8aa48a)

Using the same test script I've used for days, both cameras have been running for 23 hours with no errors. Previously, I've never seen the POE go beyond 12 hours.

wrlear, 2021-12-15:
Latest update: Since my NN inferences were running at just over 20FPS, I ran extended tests of the failing script with reduced RGB and BW camera frame rates (one at 20fps, one at 10fps). The 20FPS test reduced the average LeonOS CPU load from 93% to 73%, The 10FPS test reduced it to 36%.  I ran tests with these reductions on 2 OAK-D-PoE cameras with 2.13.2.0 sw - while running the on-board "hello world" script and displaying the RGB image. I saw NO failures in 7 days of continuous testing.

themarpe, 2021-12-15:
That's great to hear! Thanks for all the information.

There are also some stability improvements in the latest `develop`, which should resolve the script issues on PoE.

Luxonis-Brandon, 2021-12-15:
Fantastic thanks!