luxonis/depthai #553: depthai_demo.py fails with exception and told me to report to developers

diablodale, 2021-11-25:
Running `depthai_demo.py` with `face-detection-retail-0004` for 12 shaves throws exception and tells me to report to developers.

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* depthai repo at `main` currently d51f032df0a100f2879ca7318da333fb779cb6ec
* python 3.9.5 64-bit

### Repro

1. Follow instructions from https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#setup
2. activate the venv
3. Run `python depthai_demo.py`
4. see the default model run correctly
5. change model to `face-detection-retail-0004`
6. click apply restart
7. restart depthai_demo.py if the app completely exits
8. change model to `face-detection-retail-0004` again if the app exited
9. see face model working
10. change shaves to 12
11. click apply restart
12. restart depthai_demo.py if the app completely exits
13. change model to `face-detection-retail-0004` again if the app exited
14. click apply restart
15. download happens in background... then...

### Result

App throws exception. Tells me to report to developers.
Windows dialog also appears.
[image]

```
(venv) C:\njs\depthai>python depthai_demo.py
Using depthai module from:  C:\njs\depthai\venv\lib\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  2.13.0.0
Setting up demo...
Available devices:
[0] 14442C10C1A1C6D200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Stopping demo...
=== TOTAL FPS ===
[color]: 28.2
[nnInput]: 23.7
[depthRaw]: 24.9
[depth]: 24.9
[left]: 27.0
[rectifiedLeft]: 23.6
Setting up demo...
[right]: 26.4
Available devices:
[rectifiedRight]: 24.6
[nn]: 23.6
[0] 14442C10C1A1C6D200 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
Downloading C:\Users\dale\.cache\blobconverter\face-detection-retail-0004_openvino_2021.4_12shave.blob...
[==================================================]
Done
[14442C10C1A1C6D200] [15.506] [SpatialDetectionNetwork(14)] [warning] Network compiled for 12 shaves, maximum available 13, compiling for 6 shaves likely will yield in better performance
[14442C10C1A1C6D200] [15.680] [system] [critical] Fatal error. Please report to developers. Log: 'ResourceLocker' '162'
Stopping demo...
=== TOTAL FPS ===
[color]: 30.5
[left]: 0.0
[right]: 0.0
  File "C:\njs\depthai\depthai_demo.py", line 548, in run
    self.instance.run_all(self.conf)
  File "C:\njs\depthai\depthai_demo.py", line 61, in run_all
    self.run()
  File "C:\njs\depthai\depthai_demo.py", line 256, in run
    self.loop()
  File "C:\njs\depthai\depthai_demo.py", line 283, in loop
    self._pv.prepareFrames(callback=self.onNewFrame)
  File "c:\njs\depthai\depthai_sdk\src\depthai_sdk\managers\preview_manager.py", line 113, in prepareFrames
    packet = queue.tryGet()
Communication exception - possible device error/misconfiguration. Original message 'Couldn't read data from stream: 'color' (X_LINK_ERROR)'

```

Log  
[log_system_information.zip](https://github.com/luxonis/depthai/files/7604750/log_system_information.zip)

### Expected
 
No errors and 12 shaves being used.

alex-luxonis, 2021-11-25:
Thanks for the report!

The issue is related to the allocation of 2 inference threads here:
https://github.com/luxonis/depthai/blob/766ebed/depthai_sdk/src/depthai_sdk/managers/nnet_manager.py#L164
Looks like it tries to allocate 2x the number of shaves the model is compiled with.
Changing it to 1 in the code, it would work with up to 13 shaves (limited by this config with cameras, depth, spatial position calculation). 
But in fact, having 2x threads (with one NCE - neural compute engine per thread) and max 6 shaves per thread, if probably performs better than a single thread with more shaves.

@VanDavv Maybe we could make it more clear in the GUI, specifying this as number of shaves per thread, and have also the number of threads configurable.
CC also @szabi-luxonis on the firmware falling back to a single thread (with a warning) for this case.

szabi-luxonis, 2021-11-25:
The issue is not related to the above-mentioned number of threads, it's an internal resource allocation issue in firmware. Will issue a fix.

diablodale, 2021-11-25:
🤪What you said @alex-luxonis 😉

I have much to learn about the onboard capabilities of OAK: NN, hardware threads, shaves, etc. Not only the tech, but the "feel" of the tech. This demo py looks like an excellent place to experiment and learn -- hence -- me trying different shave counts to see how it affects NN fps, latency, or "feel".

The demo GUI did caution me that it probably runs better with 6 shaves. 🙉

Is your fuzz testing in place? Randomly choosing config options and then run and see what breaks. Good opportunity for automation. a few rand(), a run(), and watch for smoke. 😂

alex-luxonis, 2021-11-25:
That's true, we need to improve the testing...

>The issue is not related to the above-mentioned number of threads, it's an internal resource allocation issue in firmware when setRuntimeModeSwitch is enabled.

Ok, so the number of NN threads just indirectly affected the initial allocation for StereoDepth. I tried just with: 1x13 working, 2x6(default) working, 2x7 failing, 3x4 working, 3x5 failing.

szabi-luxonis, 2021-11-30:
@diablodale this is fixed in latest main.

diablodale, 2021-12-04:
Verified fixed with main f54806f894f940ffbd34ac9d55f157e3eb568f18
thanks!

closed 2021-12-04 by diablodale