luxonis/depthai #557: `depthai_demo.py -cnn human-pose-estimation-0001` fails with size error, then no obvious NN output

diablodale, 2021-11-29:
Running `depthai_demo.py -cnn human-pose-estimation-0001` results in a cascade of errors and/or no expected output

```
Unable to determine the nn input size. Please use --cnnInputSize flag to specify it in WxH format: -nnSize <width>x<height>
```

Running a random chosen res `depthai_demo.py -cnn human-pose-estimation-0001 -cnnsize 480x480` results in error
```
[14442C10C1A1C6D200] [40.477] [NeuralNetwork(14)] [warning] Input image (480x480) does not match NN (456x256)
```

So the system knows the size it needs.
Running `python depthai_demo.py -cnn human-pose-estimation-0001 -cnnsize 456x256` results in nothing useful shown on the human in the UI (dots, lines, boxes, labels, numbers, etc.) and nothing meaningful in stdout...
```
Received NN packet:  Mconv7_stage2_L1: (1, 38, 32, 57), Mconv7_stage2_L2: (1, 19, 32, 57)
...
```

### Expected

Running `depthai_demo.py -cnn human-pose-estimation-0001` works immediately. The demo knows download url, cc size, and how to display meaningful NN output -- perhaps similar to how the demo draws openpose.

### Log

[log_system_information.zip](https://github.com/luxonis/depthai/files/7619481/log_system_information.zip)

VanDavv, 2021-12-02:
Hi @diablodale, thanks for this report.

The demo needs `model.json` to correctly understand the network it's dealing with, so our current supported networks are listed here - https://github.com/luxonis/depthai/tree/main/resources/nn
In this case, adding just this file won't be enough, as `openpose2` uses also a custom handler to display its results (`handler.py`).
But, as you mentioned, these two networks are very similar to each other so will add `human-pose-estimation-0001` too

VanDavv, 2021-12-02:
@diablodale I just added a support for `human-pose-estimation-0001` here - https://github.com/luxonis/depthai/pull/566

closed 2021-12-02 by VanDavv

diablodale, 2021-12-03:
Verified using main f54806f894f940ffbd34ac9d55f157e3eb568f18.
Thank you. This helps me understand and interactively test the posing models :-)

PR #566 (2021-12-02): Add human-pose-estimation-0001 support
This PR introduces `human-pose-estimation-0001`, providing similar output as `openpose2` network. Resolves #557