luxonis/depthai #562: regression: windows `python install_requirements.py` fails around pyqt5

diablodale, 2021-12-01:
There has been a regression on Windows for install requirements. Running `python install_requirements.py` ...
Works on commit d51f032df0a100f2879ca7318da333fb779cb6ec which was used in issue #553
Fails on current main e76bac031e62bb3ea31bcc87a52365388c85243a

### Setup

* Microsoft Windows [Version 10.0.19044.1348]
* Python 3.9.5
* depthai commit e76bac031e62bb3ea31bcc87a52365388c85243a, the commit before it b1747a9f392daf48403c65ae1e40a2f54ae5093c also fails

### Repro

1. Follow instructions from https://docs.luxonis.com/en/latest/pages/tutorials/first_steps/#setup
2. When you run `python install_requirements.py` you get an unrecoverable error

### Results

```
(.venv) C:\njs\depthai>python install_requirements.py
pip 21.3.1 from C:\njs\depthai\.venv\lib\site-packages\pip (python 3.9)
Requirement already satisfied: pip in c:\njs\depthai\.venv\lib\site-packages (21.3.1)
WARNING: Skipping opencv-python as it is not installed.
WARNING: Skipping opencv-contrib-python as it is not installed.
WARNING: Skipping depthai as it is not installed.
Looking in indexes: https://pypi.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring opencv-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Obtaining file:///C:/njs/depthai/depthai_sdk (from -r requirements.txt (line 10))
  Preparing metadata (setup.py) ... done
Collecting requests==2.26.0
  Using cached requests-2.26.0-py2.py3-none-any.whl (62 kB)
Collecting pyrebase4==4.5
  Using cached Pyrebase4-4.5.0-py3-none-any.whl (8.9 kB)
Collecting argcomplete==1.12.1
  Using cached argcomplete-1.12.1-py2.py3-none-any.whl (38 kB)
Collecting opencv-python==4.5.1.48
  Using cached opencv_python-4.5.1.48-cp39-cp39-win_amd64.whl (34.9 MB)
Collecting opencv-contrib-python==4.5.1.48
  Using cached opencv_contrib_python-4.5.1.48-cp39-cp39-win_amd64.whl (41.2 MB)
Collecting pyqt5==5.14.2
  Using cached PyQt5-5.14.2.tar.gz (3.2 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: 'C:\njs\depthai\.venv\Scripts\python.exe' 'C:\njs\depthai\.venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' prepare_metadata_for_build_wheel 'C:\Users\dale\AppData\Local\Temp\tmp20vbo1hq'
       cwd: C:\Users\dale\AppData\Local\Temp\pip-install-rp77wfxe\pyqt5_78c07efc06734daeb2837e56a3be9e3a
  Complete output (29 lines):
  Traceback (most recent call last):
    File "C:\njs\depthai\.venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 156, in prepare_metadata_for_build_wheel
      hook = backend.prepare_metadata_for_build_wheel
  AttributeError: module 'sipbuild.api' has no attribute 'prepare_metadata_for_build_wheel'

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "C:\njs\depthai\.venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 363, in <module>
      main()
    File "C:\njs\depthai\.venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 345, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "C:\njs\depthai\.venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py", line 160, in prepare_metadata_for_build_wheel
      whl_basename = backend.build_wheel(metadata_directory, config_settings)
    File "C:\Users\dale\AppData\Local\Temp\pip-build-env-vws6ar58\overlay\Lib\site-packages\sipbuild\api.py", line 51, in build_wheel
      project = AbstractProject.bootstrap('pep517')
    File "C:\Users\dale\AppData\Local\Temp\pip-build-env-vws6ar58\overlay\Lib\site-packages\sipbuild\abstract_project.py", line 83, in bootstrap
      project.setup(pyproject, tool, tool_description)
    File "C:\Users\dale\AppData\Local\Temp\pip-build-env-vws6ar58\overlay\Lib\site-packages\sipbuild\project.py", line 479, in setup
      self.apply_user_defaults(tool)
    File "C:\Users\dale\AppData\Local\Temp\pip-install-rp77wfxe\pyqt5_78c07efc06734daeb2837e56a3be9e3a\project.py", line 62, in apply_user_defaults
      super().apply_user_defaults(tool)
    File "C:\Users\dale\AppData\Local\Temp\pip-build-env-vws6ar58\overlay\Lib\site-packages\pyqtbuild\project.py", line 70, in apply_user_defaults
      super().apply_user_defaults(tool)
    File "C:\Users\dale\AppData\Local\Temp\pip-build-env-vws6ar58\overlay\Lib\site-packages\sipbuild\project.py", line 225, in apply_user_defaults
      self.builder.apply_user_defaults(tool)
    File "C:\Users\dale\AppData\Local\Temp\pip-build-env-vws6ar58\overlay\Lib\site-packages\pyqtbuild\builder.py", line 66, in apply_user_defaults
      raise PyProjectOptionException('qmake',
  sipbuild.pyproject.PyProjectOptionException
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/4d/81/b9a66a28fb9a7bbeb60e266f06ebc4703e7e42b99e3609bf1b58ddd232b9/PyQt5-5.14.2.tar.gz#sha256=bd230c6fd699eabf1ceb51e13a8b79b74c00a80272c622427b80141a22269eb0 (from https://pypi.org/simple/pyqt5/) (requires-python:>=3.5). Command errored out with exit status 1: 'C:\njs\depthai\.venv\Scripts\python.exe' 'C:\njs\depthai\.venv\lib\site-packages\pip\_vendor\pep517\in_process\_in_process.py' prepare_metadata_for_build_wheel 'C:\Users\dale\AppData\Local\Temp\tmp20vbo1hq' Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pyqt5==5.14.2 (from versions: 5.12.3, 5.14.0, 5.14.1, 5.14.2, 5.15.0, 5.15.1, 5.15.2, 5.15.3, 5.15.4, 5.15.5, 5.15.6)
ERROR: No matching distribution found for pyqt5==5.14.2
Traceback (most recent call last):
  File "C:\njs\depthai\install_requirements.py", line 45, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"])
  File "c:\python39\lib\subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['C:\\njs\\depthai\\.venv\\Scripts\\python.exe', '-m', 'pip', 'install', '-U', '--prefer-binary', '-r', 'requirements.txt']' returned non-zero exit status 1.
```

### Expected

No error. Installs requirements cleanly

### Logs

This log is from a scenario after the install requirements failed somewhere during its work.
[log_system_information.zip](https://github.com/luxonis/depthai/files/7636179/log_system_information.zip)

VanDavv, 2021-12-02:
Hi @diablodale, thanks for reporting the issue!
I pushed a fix in this PR - https://github.com/luxonis/depthai/pull/564, should be merged soon

closed 2021-12-02 by VanDavv

diablodale, 2021-12-03:
🙂Verified, looks good to me at main f54806f894f940ffbd34ac9d55f157e3eb568f18
thank you

PR #564 (2021-12-02): use better requirements specification for PyQt5
This PR introduces less strict version specification for PyQt5, requiring just PyQt5 5.x. Resolves #562