luxonis/depthai #567: [Feature-Request] {Add support for aligning the depth/disparity confidence map with the RGB image}

nrgsy, 2021-12-02:
### Start with the `why`:
For the OAK-D camera, it is currently possible to align depth with the RGB image, but it is not possible to also align the confidence map with the RGB image. It would be useful to have the both the depth and the confidence for every pixel in the RGB image because otherwise we don't know how much to trust each depth value when processing depth images on the host.

An application for this is in building colorized 3D occupancy grid maps - the per-pixel confidence can be used to weight how much each voxel update should be trusted.

Another application is in assigning uncertainty to detected objects - per-pixel confidence can be used to put error bars and the position of a detected object, which can influence the behavior of a robot.

I understand it is currently possible to remove depth values that are associated with a confidence beyond a given threshold, but this isn't a granular enough control for the aforementioned applications. 

### Move to the `what`:
When this is in the pipeline setup:
```
    stereo.setDepthAlign(depthai.CameraBoardSocket.RGB)
```
The retrieved depth images are aligned resized to match the RGB image, even if the monocular cameras are running at a lower resolution. I propose that a `setConfidenceAlign` setting be added that would do the same thing to the confidenceMap. So confidence maps would be output at same resolution as the RGB image and aligned with it. The relevant part of the pipeline setup might then look something like this:
```
    stereo = pipeline.create(depthai.node.StereoDepth)
    stereo.setConfidenceAlign(depthai.CameraBoardSocket.RGB)
    confidence_out = pipeline.create(depthai.node.XLinkOut)
    confidence_out.setStreamName('confidence')
    stereo.confidenceMap.link(confidence_out.input)
```

### Move to the `how`:
Not 100% sure the best way to implement it, but perhaps the `DepthAlignment` class here could be refactored/generalized such that it would work for both confidence and depth images:
https://github.com/luxonis/depthai-core/blob/84fe1e164d6b505cc00b8d82dacbaaf8026fc24d/example/main.cpp#L15

This would be a huge help to the folks and Built Robotics and I'm sure others in the robotics community would appreciate the flexibility that the granular per-pixel control would enable.

Luxonis-Brandon, 2021-12-03:
Great feature idea!  Thank you for writing this up.  I see that Szabi already snagged this one.  So I suspect this won't be long (assuming there isn't some unforeseen technical limit).  :-)

szabi-luxonis, 2021-12-03:
> Great feature idea! Thank you for writing this up. I see that Szabi already snagged this one. So I suspect this won't be long (assuming there isn't some unforeseen technical limit). :-)

Relatively straightforward, but at the moment I'm focusing on higher priority tasks. Just self-assigned to not forget.
Will keep you updated once it's started/done @nrgsy .

Luxonis-Brandon, 2021-12-03:
Thanks, Szabi!  And please let us know if there are any drop-dead like deadlines or needs on this @nrgsy .

szabi-luxonis, 2022-01-16:
Support added in this PR: https://github.com/luxonis/depthai-python/pull/488
@nrgsy

closed 2022-01-26 by szabi-luxonis