luxonis/depthai #568: [BUG] {Error running python3 depthai_demo.py}

JaCoHSU, 2021-12-03:
Hey guys, 

My Oak-D Lite just arrived. I wanted to test it with depthai. What I did is this:

```
git clone https://github.com/luxonis/depthai.git
cd depthai
python3 install_requirements.py
```

Installation seemed to work without errors but when I executed this:
`python3 depthai_demo.py`

I get the following error. Does anyone know how to fix it or why this error occured?

```
 python3 depthai_demo.py
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Using depthai module from:  /home/falke/.local/lib/python3.6/site-packages/depthai.cpython-36m-x86_64-linux-gnu.so
Depthai version installed:  2.13.3.0
QObject::moveToThread: Current thread (0x37e0aa0) is not the object's thread (0x3795bd0).
Cannot move to target thread (0x37e0aa0)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/falke/.local/lib/python3.6/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl.

Aborted (core dumped)
```

alex-luxonis, 2021-12-03:
@JaCoHSU It's possible some packages need to be installed for Qt on Linux, these added here:
https://github.com/luxonis/depthai-python/commit/79f47c7c5e4dc55fcafdece9cca684b12702e03f

You can install with `apt install`, or just run the command to install all dependencies:
```
sudo wget -qO- https://docs.luxonis.com/install_dependencies.sh | bash
```
(from https://docs.luxonis.com/projects/api/en/latest/install/?#ubuntu)

JaCoHSU, 2021-12-03:
@alex-luxonis thanks for the quick response.

I ran the install_dependencies.sh

```
0 upgraded, 0 newly installed, 0 to remove and 26 not upgraded.
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in ./.local/lib/python3.6/site-packages (21.3.1)
SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"
```

I also did run the install_requierments.py again:

```
pip 21.3.1 from /home/falke/.local/lib/python3.6/site-packages/pip (python 3.6)
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Requirement already satisfied: pip in /home/falke/.local/lib/python3.6/site-packages (21.3.1)
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Found existing installation: opencv-python 4.5.1.48
Uninstalling opencv-python-4.5.1.48:
  Successfully uninstalled opencv-python-4.5.1.48
Found existing installation: opencv-contrib-python 4.5.1.48
Uninstalling opencv-contrib-python-4.5.1.48:
  Successfully uninstalled opencv-contrib-python-4.5.1.48
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Found existing installation: depthai 2.13.3.0
Uninstalling depthai-2.13.3.0:
  Successfully uninstalled depthai-2.13.3.0
/usr/lib/python3/dist-packages/secretstorage/dhcrypto.py:15: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
/usr/lib/python3/dist-packages/secretstorage/util.py:19: CryptographyDeprecationWarning: int_from_bytes is deprecated, use int.from_bytes instead
  from cryptography.utils import int_from_bytes
Looking in indexes: https://pypi.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring opencv-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Obtaining file:///home/falke/installations/depthai/depthai_sdk (from -r requirements.txt (line 10))
  Preparing metadata (setup.py) ... done
Requirement already satisfied: requests==2.26.0 in /home/falke/.local/lib/python3.6/site-packages (from -r requirements.txt (line 1)) (2.26.0)
Requirement already satisfied: pyrebase4==4.5 in /home/falke/.local/lib/python3.6/site-packages (from -r requirements.txt (line 2)) (4.5.0)
Requirement already satisfied: argcomplete==1.12.1 in /home/falke/.local/lib/python3.6/site-packages (from -r requirements.txt (line 3)) (1.12.1)
Collecting opencv-python==4.5.1.48
  Using cached opencv_python-4.5.1.48-cp36-cp36m-manylinux2014_x86_64.whl (50.4 MB)
Collecting opencv-contrib-python==4.5.1.48
  Using cached opencv_contrib_python-4.5.1.48-cp36-cp36m-manylinux2014_x86_64.whl (56.3 MB)
Requirement already satisfied: pyqt5<6,>5 in /home/falke/.local/lib/python3.6/site-packages (from -r requirements.txt (line 12)) (5.15.6)
Collecting depthai==2.13.3.0
  Using cached depthai-2.13.3.0-cp36-cp36m-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (9.7 MB)
Requirement already satisfied: idna<4,>=2.5 in /home/falke/.local/lib/python3.6/site-packages (from requests==2.26.0->-r requirements.txt (line 1)) (2.10)
Requirement already satisfied: certifi>=2017.4.17 in /home/falke/.local/lib/python3.6/site-packages (from requests==2.26.0->-r requirements.txt (line 1)) (2020.6.20)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/falke/.local/lib/python3.6/site-packages (from requests==2.26.0->-r requirements.txt (line 1)) (1.25.10)
[... cut ...]
Requirement already satisfied: wcwidth in /home/falke/.local/lib/python3.6/site-packages (from prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0->ipython>=4.0.0->ipywidgets->open3d==0.10.0.0->-r requirements-optional.txt (line 1)) (0.2.5)
Requirement already satisfied: webencodings in /usr/lib/python3/dist-packages (from bleach->nbconvert->notebook->open3d==0.10.0.0->-r requirements-optional.txt (line 1)) (0.5)
Requirement already satisfied: packaging in /home/falke/.local/lib/python3.6/site-packages (from bleach->nbconvert->notebook->open3d==0.10.0.0->-r requirements-optional.txt (line 1)) (20.4)
```

and when i run python3 depthai_demo.py I get the same error as posted before. Any idea?

alex-luxonis, 2021-12-03:
What OS is this on, Ubuntu 18.04? I see there Python 3.6 (default on 18.04).

It's possible to run also without the Qt GUI, with the old OpenCV interface, but most of the options there need to be set from command line:
`python3 depthai_demo.py -gt cv`

JaCoHSU, 2021-12-03:
yes, its 18.04

`python3 depthai_demo.py -gt cv ` does work. I guess I need to live with it for now. Thanks!

VanDavv, 2021-12-03:
Hi @JaCoHSU , 

I installed Ubuntu 18.04 and I was able to make it work on this PR - https://github.com/luxonis/depthai/pull/571. The root cause of the issue is the QT library conflict between `opencv-python` and `PyQt5`, like mentioned [in this issue on Qt forum](https://forum.qt.io/topic/119109/using-pyqt5-with-opencv-python-cv2-causes-error-could-not-load-qt-platform-plugin-xcb-even-though-it-was-found). 

To make the Qt GUI work on Ubuntu 18.04 please use the following commands

```
$ cd depthai  # repo root
$ git pull
$ git checkout ubuntu_1804_fix
$ export LD_LIBRARY_PATH="$HOME/.local/lib/python3.6/site-packages/PyQt5/Qt5/lib/
$ python3 depthai_demo.py
```

Will check if I can work around setting LD_LIBRARY_PATH, so that the script can load everything on its own.

piponazo, 2021-12-05:
Hi. I have the same problem under Ubuntu 20.04:
```
(oak-d) luis@ryzenLinux:/media/linuxDev/Oak-D/depthai(main)$ python3 depthai_demo.py 
Using depthai module from:  /media/linuxDev/envs/oak-d/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
Depthai version installed:  2.13.3.0
QObject::moveToThread: Current thread (0x22d2060) is not the object's thread (0x21fbc70).
Cannot move to target thread (0x22d2060)

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/media/linuxDev/envs/oak-d/lib/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl.

Aborted (core dumped)
```
Running with the CV GUI works fine (`python3 depthai_demo.py -gt cv`)

Luxonis-Brandon, 2021-12-05:
Thanks for the report.  Investigating.

JaCoHSU, 2021-12-06:
Hey @VanDavv 
your suggestion worked for me. The gui opens :) Some core dumbs occur from time to time, but other than that it seems to work. thanks!

VanDavv, 2021-12-21:
@JaCoHSU happy that it helped!

@piponazo could you try again with the latest main? I added the supervisor mode that should now set these flags automatically - #573

barney2074, 2022-04-29:
Hi @VanDavv 

I am also having this problem- (I've got an 18.04 system for compatibility with other applications)
It lloks like the fix has been removed from git ??- I get the following error

thanks

Andrew

```
ubuntu9@ubuntu9-HP-Z620-Workstation:~/depthai$ git pull
Already up to date.
ubuntu9@ubuntu9-HP-Z620-Workstation:~/depthai$ git checkout ubuntu_1804_fix
error: pathspec 'ubuntu_1804_fix' did not match any file(s) known to git.
ubuntu9@ubuntu9-HP-Z620-Workstation:~/depthai$ 
```

alex-luxonis, 2022-04-29:
Yeah, the branch was deleted as cleanup, it was merged in the meantime, in PR https://github.com/luxonis/depthai/pull/571.
Could you post the logs (terminal prints) running with latest `main`?

barney2074, 2022-04-29:
Hi @alex-luxonis 

Is this what you mean ??

I get the 'color' & 'depth' windows, just not the main GUI

```
ubuntu9@ubuntu9-HP-Z620-Workstation:~/depthai$ source depthai/bin/activate
(depthai) ubuntu9@ubuntu9-HP-Z620-Workstation:~/depthai$ python3 depthai_demo.pyUsing depthai module from:  /home/ubuntu9/depthai/depthai/lib/python3.9/site-packages/depthai.cpython-39-x86_64-linux-gnu.so
Depthai version installed:  2.15.3.0.dev+7b7cd9224b0a31c684231e880beff9ae38bade6c
Traceback (most recent call last):
  File "/home/ubuntu9/depthai/depthai_demo.py", line 1079, in <module>
    runQt()
  File "/home/ubuntu9/depthai/depthai_demo.py", line 604, in runQt
    from gui.main import DemoQtGui
  File "/home/ubuntu9/depthai/gui/main.py", line 7, in <module>
    from PyQt5.QtQml import QQmlApplicationEngine, qmlRegisterType, qmlRegisterSingletonType, QQmlEngine
ImportError: /usr/lib/x86_64-linux-gnu/libgssapi_krb5.so.2: symbol krb5_ser_context_init version krb5_3_MIT not defined in file libkrb5.so.3 with link time reference
Sentry is attempting to send 2 pending error messages
Waiting up to 2 seconds
Press Ctrl-C to quit
Error while running demo script... Command '"/home/ubuntu9/depthai/depthai/bin/python3" depthai_demo.py --noSupervisor --guiType qt' returned non-zero exit status 1.
Waiting 5s for the device to be discoverable again...
Setting up demo...
Available devices:
[0] 18443010F160F50800 [X_LINK_UNBOOTED]
USB Connection speed: UsbSpeed.SUPER
QObject::moveToThread: Current thread (0x562ad63ebb70) is not the object's thread (0x562ad7241980).
Cannot move to target thread (0x562ad63ebb70)

QObject::moveToThread: Current thread (0x562ad63ebb70) is not the object's thread (0x562ad7241980).
Cannot move to target thread (0x562ad63ebb70)

QObject::moveToThread: Current thread (0x562ad63ebb70) is not the object's thread (0x562ad7241980).
Cannot move to target thread (0x562ad63ebb70)

QObject::moveToThread: Current thread (0x562ad63ebb70) is not the object's thread (0x562ad7241980).
Cannot move to target thread (0x562ad63ebb70)
```

VincentLombardi, 2022-05-05:
I am having the exact same issue as you @barney2074. Fresh Ubuntu 22.04 VM on VMware workstation. Used the install requirements script. The camera connects just fine and the demo starts but I only get the color and depth windows. When I run natively on Windows 10 I get the full GUI just fine

Luxonis-Brandon, 2022-05-10:
Oh yes sorry about the delay.  So this means the QT GUI has failed.  And when that happens, the demo automatically falls back to the OpenCV GUI.

We're in the process of re-writing the demo to use a web GUI instead.  ETA probably 3-4 months.  For now, if the QT GUI doesn't show up, the OpenCV GUI can be used with command line arguments to specify.  There's also an experimental webGUI, but we're abandoning that for an alternate approach.  

Thoughts?

Sorry again about the delay.

barney2074, 2022-05-10:
Hi Brandon

thanks for the response
I've got it working (it is a little intermittent) on a 20.04 system, so I'm fine for now & will look forward to the webGUI

thanks

Andrew

Luxonis-Brandon, 2022-05-11:
Great thank you!

barney2074, 2022-05-17:
Hi Vincent

I'm not sure- I started with a fresh 20.04.4 LTS (note- not **22.**04) build

I don't know much about QT & if it is reliant on specific graphics drivers- but went through some real headaches with NVIDIA drivers, CUDA, cuDNN etc for other reasons

Andrew

VincentLombardi, 2022-05-17:
Hi Andrew,

Awesome! Thanks for the info. I will give it a try. 

Appreciate your help,
Vincent