luxonis/depthai #575: Capturing full 4:3 4056x3040 frames to be inferenced on.

justin-larking, 2021-12-07:
Would like to be able to inference and view the full view of the camera in 4:3 to provide a larger view area than a normal 4k image cropped from the centre of the original resolution. 

Ability to set video size to the full 12.3MP resolution of the camera and get a 4:3 aspect ratio image out along with being passed to do inferencing on.

i.e. cam_rgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_12_MP)
and cam_rgb.setvideosize(4056, 3040)

Luxonis-Brandon, 2021-12-07:
Thanks.  If I'm understanding correctly I think @Erol444 has a reference for how to accomplish this.  I looked but couldn't immediately find it.

Erol444, 2021-12-08:
Hello @justin-larking , from my understanding of ISP, only 12MP `still` frames and `isp` output will have the actual sensor full FOV - so 4056x3040. For some reason, isp is actually `4032x3040`, might be something to do with HW limitation (eg width must be multiplies of 32/64).
So I would use `isp` output from ColorCamera node, and you need to convert the YUV420 to RGB (+ downscale), will write a simple demo for that asap.

alex-luxonis, 2021-12-08:
Full resolution (4056 width) should work with `isp` in this example:
https://github.com/luxonis/depthai-experiments/blob/f7542b9/gen2-color-isp-raw/main.py
`python3 main.py -res 12mp`

But when encoding, e.g JPEG with `still` output of ColorCamera, the resolution is slightly reduced to a multiple of 32, due to VideoEncoder hardware limitations.

Erol444, 2021-12-08:
@justin-larking  Full fov NN inferencing demo here:
https://github.com/luxonis/depthai-experiments/tree/full_fov/gen2-full-fov-nn

justin-larking, 2021-12-08:
After a quick mockup of that code:
```
import cv2
import depthai as dai
import time

# Start defining a pipeline
pipeline = dai.Pipeline()
print("Creating Color Camera...")
cam_rgb = pipeline.createColorCamera()
cam_rgb.setInterleaved(False)
cam_rgb.setResolution(dai.ColorCameraProperties.SensorResolution.THE_12_MP)
cam_rgb.setBoardSocket(dai.CameraBoardSocket.RGB)
cam_rgb.setStillSize(4032, 3040)
cam_rgb.setColorOrder(dai.ColorCameraProperties.ColorOrder.RGB)

controlIn = pipeline.create(dai.node.XLinkIn)

cam_xout = pipeline.createXLinkOut()
cam_xout.setStreamName("still")
cam_xout.input.setBlocking(False)
cam_xout.input.setQueueSize(1)

controlIn.setStreamName('control')

controlIn.out.link(cam_rgb.inputControl)
cam_rgb.still.link(cam_xout.input)

with dai.Device(pipeline) as device:
    still = device.getOutputQueue(name="still", maxSize=1, blocking=False)
    controlQueue = device.getInputQueue('control')

    prev_frame_time = 0
    new_frame_time = 0

    ctrl = dai.CameraControl()
    while True:
        new_frame_time = time.time()
        ctrl.setCaptureStill(True)
        controlQueue.send(ctrl)
        print(ctrl.getCaptureStill())
        frameIn = still.get()
        print(frameIn)
        frame = frameIn.getCvFrame()
        fps = 1/(new_frame_time-prev_frame_time)
        prev_frame_time = new_frame_time
        cv2.putText(frame, str(int(fps)), (900 ,50), cv2.FONT_HERSHEY_SIMPLEX, 2, (255,255,255), 4, 2)
        cv2.imshow("frame", frame)
        if cv2.waitKey(1) == ord("c"):
            print("Saving")
            cv2.imwrite("frame"+str(1)+".jpg", frame)
            print("frame"+str(1)+".jpg")
        if cv2.waitKey(1) == ord('q'):
            break
```
I have some questions. Is it possible to obtain full fov images but at a lower resolutions, as for full fov sesnor resolution must be set to 12MP but how other than resizing can stills be captured in lower res. I ask as this code runs at 2-3fps, what i'm ultimately trying to achieve is full fov at a video frame rate.

alex-luxonis, 2021-12-08:
@justin-larking Yes, it's possible to downscale, as for example in the code @Erol444 posted, using `.setIspScale(1, 5)` for a 1/5 scaling of the `isp` output, that is `(4056, 3040) * 1/5 = (811.2, 608) -> (812, 608)`
(actual formula for the scaled size: https://github.com/luxonis/depthai-core/blob/6620089/src/pipeline/node/ColorCamera.cpp#L318)

Also possible to try it with [gen2-color-isp-raw](https://github.com/luxonis/depthai-experiments/blob/f7542b9/gen2-color-isp-raw/main.py): `python3 main.py -res 12mp -ds 5`

The `isp` output is a source for `preview` / `video` / `still`, see more details here:
https://docs.luxonis.com/projects/api/en/latest/components/nodes/color_camera/

justin-larking, 2021-12-08:
Ah I see now, I shouldn't have skipped over that tutorial on isp. I'll close the this as its possible and relatively easy to do.

closed 2021-12-08 by justin-larking